meshtastic/Meshtastic-Apple #257: [Feature Request]: To open region Russia for AppStore

debitorka, 2022-12-07:
### OS

iOS

### Description

Region Russia unavailable now in AppStore
Please, open this region, if its possible

ANS-spb, 2022-12-08:
I need it too

garthvh, 2022-12-09:
Sorry, not possible

Anonym-tsk, 2022-12-09:
> Sorry, not possible

What reason?

closed 2022-12-12 by garthvh

sermelipharo, 2024-07-16:
Still required. Governmental censorship in Russia is rising up, and VPNs are already banned

Leka1B, 2025-02-26:
It's needed more than ever with governments just up and cutting off the internet in some regions.

This would be the best available solution to governments controlling communication

Flumeded, 2025-04-03:
> Sorry, not possible

If you need any contribution from my side, just let us know.
I am sure we can tackle the roadblocks you are facing

Nenormalniy, 2025-07-09:
Colleagues, we turned off somewhere. We make a technology, not a political statement. Any of you can be in any emergency, in any country. We must try to spread such technology wherever possible.
What is the reason you made the application inaccessible in Russia? For what? To what?