meshtastic/Meshtastic-Apple #264: [Feature Request]: Pop-up upon reaching duty cycle limit

GUVWAF, 2022-12-13:
### OS

iOS, iPadOS, macOS

### Description

In the rare case that someone exceeds the duty cycle, from 2.0.7 on the device sends a NAK with "Routing_Error_DUTY_CYCLE_LIMIT" to the client. At this point it would be good to show a pop-up telling the user why it failed and to try it again in a couple of minutes. And maybe giving the option to override it, with warning that you then may violate the local regulations. 

BTW, I had the chance to try the app on an iPad mini last weekend and I really liked it! It looks really polished and I find it very intuitive to use.

garthvh, 2022-12-13:
Is the error not in the routing error enum in the protobufs?

GUVWAF, 2022-12-13:
It is: https://github.com/meshtastic/protobufs/blob/master/mesh.proto#L511

garthvh, 2022-12-13:
I have to add it to my enum, but once that is in it will show up under a message like the other ack errors. It will also show up in the mesh log for other packet types

garthvh, 2022-12-28:
Uses existing ACK error display in messaging, will also display in the mesh log

In testflight, will release with 2.0.9

closed 2022-12-28 by garthvh