meshtastic/Meshtastic-Apple #277: [Feature Request]: Distinguish between implicit ACK and ‘real’ ACK for DMs

GUVWAF, 2022-12-29:
### OS

iOS, iPadOS, macOS

### Description

From firmware 2.0.9 on, ‘real’ ACKs (which are sent only for DMs) will be flooded back to the original sender. In this way, we can guarantee that a DM was received at the intended receiver. It would be nice to show whether an implicit ACK (meaning it was repeated on the mesh) or a real ACK was received. 

When a client gets an ACK, it then needs to check the ‘from’ field. If it is your own nodeNum, it is a rebroadcast (implicit ACK), if it is the nodeNum of the intended receiver, then the reception is confirmed.

garthvh, 2022-12-29:
Will I get a max retransmission error if the real ack is not returned?

GUVWAF, 2022-12-29:
No, that logic hasn't changed. You will get this only when even no implicit ACK was received after three retransmissions. 
Since it might need multiple hops to get to the destination, it is hard to predict when the real ACK should have been returned. So we only have zero-hop reliability.

garthvh, 2022-12-31:
This is in TestFlight and and go out with 2.0.10

An implicit ACK on a dm shows the Acknowledged text under the message in orange, it changes to grey if the real Ack arrives from the destination node.

There is also information and the timestamp in the Message Details section on the message context menu that comes up when you long press on a message.

closed 2022-12-31 by garthvh