meshtastic/Meshtastic-Apple #296: [Feature Request]: WiFi connection via TCP to Meshtastic

tb0hdan, 2023-01-25:
### OS

iOS

### Description

Hi,

It is not possible to connect to Meshtastic via WiFi from iOS, but Android has such feature.

APP version: 2.0.10

garthvh, 2023-01-25:
I have done some refactoring to support multiple connection types, but it is pretty far in the backlog.  Maybe will look at it again in a year or if the state of ESP32 wifi improves.

ut3uku, 2023-04-28:
It is a very important function of the app. I am waiting for it

garthvh, 2023-07-09:
Closing in favor of https://github.com/meshtastic/Meshtastic-Apple/issues/368

geeksville, 2023-09-01:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/macos-app-ip-support/8100/4

PatrickHuetter, 2024-02-25:
+1 

Another option could be to develop a ionic app with some parts native. That would decrease the development efforts. The typescript services could also be re-used in the web based Meshtastic client.

garthvh, 2024-03-17:
I estimate it will require 25K in dev time minimum to refactor the whole code base and add this feature, have made this the first feature with a bounty available.

AssetBurned, 2024-03-23:
for people following https://github.com/meshtastic/Meshtastic-Apple/issues/506 and https://github.com/meshtastic/Meshtastic-Apple/issues/550 and not using discourse.

basementbuilder17, 2024-04-07:
i am also looking for this feature. Only access to iOS, really would like to manage all my nodes from there.

powersjcb, 2024-09-07:
I don't see this documented anywhere, but one additional reason this feature would be particularly helpful would be to enable dev work using iOS Simulator. Lack of support for Simulator requires anyone wanting to work on this to setup the appropriate Apple Developer Membership so they can self-sign the application for use on their own phone. (costs $100 per year) 

Being able to develop the app over a local network would be a nice to have, but at the same time there's probably some positive tradeoffs of devs using the exact setup clients will utilize. (BLE)

Definitely concur with @garthvh that this is a "large" rework. (its at least needing to touch 3-4 layers of abstractions)

Incase anyone else is interested, I'm attaching some links to the `Meshtastic-Android` source that already implements this.
* [BluetoothInterface.kt](https://github.com/meshtastic/Meshtastic-Android/blob/716b75c6bed9400623609e990a6b3e11e1a7d707/app/src/main/java/com/geeksville/mesh/repository/radio/BluetoothInterface.kt)
* [TCPInterface.kt](https://github.com/meshtastic/Meshtastic-Android/blob/716b75c6bed9400623609e990a6b3e11e1a7d707/app/src/main/java/com/geeksville/mesh/repository/radio/TCPInterface.kt)
* [RadioInterfaceService.kt](https://github.com/meshtastic/Meshtastic-Android/blob/716b75c6bed9400623609e990a6b3e11e1a7d707/app/src/main/java/com/geeksville/mesh/repository/radio/RadioInterfaceService.kt)
* [StreamInterfaceService.kt](https://github.com/meshtastic/Meshtastic-Android/blob/716b75c6bed9400623609e990a6b3e11e1a7d707/app/src/main/java/com/geeksville/mesh/repository/radio/StreamInterface.kt#L7) - used for TCP and serial

Hude06, 2024-10-08:
I'm down to work on a type script Capacitor JS (The new Ionic) app for it, Something similar to the flutter multi mesh app but try to reuse some of the web version. Is there a JS api already?

garthvh, 2024-10-08:
Entirely swiftui

drewzh, 2024-10-08:
> I'm down to work on a type script Capacitor JS (The new Ionic) app for it, Something similar to the flutter multi mesh app but try to reuse some of the web version. Is there a JS api already?

There is and I'm already working on a PWA/Capictor Angular app. Already have nodes list, map and messaging.

cpatulea, 2024-10-12:
[New to iOS development, new to Meshtastic - I might say something stupid] How would iOS TCP connection work? As far as I know, iOS is very aggressive in putting apps to sleep, so long lived TCP connections almost never work. For example it's almost impossible to run an IRC client natively on iOS, it's almost always through a server side proxy.

Maybe this could work using HTTP REST (like the web client) and the Meshtastic device buffering messages until iOS wakes up. But notifications could never work, so user would have to "manually open the app" to receive any messages.

drewzh, 2024-10-13:
> Hey, Drewzh is it on github? I would love to check it out @drewzh 

I'll share when the time is right. It's on GitHub, but private. Once I get things working smoothly I'll make it public. I don't like pushing anything half finished or broken, which many things currently are.

tannewt, 2024-12-25:
Looks like this may work: https://developer.apple.com/documentation/networkextension/receiving-voice-and-text-communications-on-a-local-network

A provider is allowed to run a long lived TCP connection when on an allowed WiFi (or lte) network. It then notifies the app of incoming messages or calls.

garthvh, 2024-12-25:
> Looks like this may work: https://developer.apple.com/documentation/networkextension/receiving-voice-and-text-communications-on-a-local-network
> 
> A provider is allowed to run a long lived TCP connection when on an allowed WiFi (or lte) network. It then notifies the app of incoming messages or calls.

I have looked at this, as you can see in the sample it requires a server

tannewt, 2024-12-25:
Wouldn't the esp32 Meshtastic device be the server? It looks like the NEAppPushProvider works with any tcp connection.

garthvh, 2024-12-26:
> Wouldn't the esp32 Meshtastic device be the server? It looks like the NEAppPushProvider works with any tcp connection.

Self signed certificate does not work, this is a dead end.

tannewt, 2024-12-26:
Did you install the root cert? https://jozefcipa.com/blog/self-signed-ssl-certificates-on-ios/

If you don't have time to do this I understand. I'm just trying to show it is feasible. I'd love to have a home node that works on my WiFi network anywhere in the house.

garthvh, 2024-12-26:
> Did you install the root cert? https://jozefcipa.com/blog/self-signed-ssl-certificates-on-ios/
> 
> If you don't have time to do this I understand. I'm just trying to show it is feasible. I'd love to have a home node that works on my WiFi network anywhere in the house.

Not a reasonable solution with as many app users as meshtastic has.

tannewt, 2025-01-02:
Don't Meshtastic devices make the HTTP API available without TLS?

mbenitog, 2025-05-06:
> Long running tcp connections are not possible on iOS this feature is not coming unless this changes.

We don’t need network connections to be long running. Most users wanting to use wifi will just open the app to check every once in a while. We just need one of the most basic features to just exist 🙂

I don’t really see why not, reason has changed several times. All I’ve read here is weird pushback against implementing this feature.

garthvh, 2025-05-07:
> We don’t need network connections to be long running. Most users wanting to use wifi will just open the app to check every once in a while. We just need one of the most basic features to just exist [...]

This feature is not possible on iOS, the answer has been the same every time.