meshtastic/Meshtastic-Apple #308: [Feature Request]: Client Side Weather Station Conditions Values

garthvh, 2023-02-05:
### OS

iOS, iPadOS, macOS

### Description

I want to extrapolate a few weather stations values from locally available sensors if telemetry modules are available on a device.

* [ ] Simple current conditions icon like the WeatherKid currentConditions with basic condition types we can calculate from a BME680
* [ ] Indoor Air Quality for BME680
* [ ] Air Quality Index calculation for more advanced sensors
* [ ] Current Temperature
* [ ] AQI line display like is available in Maps

asjmcguire, 2024-03-31:
dew point = Tc - ((100 - RH)/5.)
Dew point = Temp in celcius - (( 100 - humidity) /5)

Dew point is a better calculation for snow, because the dew point needs to be close to freezing. The humidity does not need to be anywhere close to 98% for snow.

garthvh, 2024-08-03:
> dew point = Tc - ((100 - RH)/5.) Dew point = Temp in celcius - (( 100 - humidity) /5)
> 
> Dew point is a better calculation for snow, because the dew point needs to be close to freezing. The humidity does not need to be anywhere close to 98% for snow.

Wound up using the Magnus formula as it seems to get me a little closer to what I was seeing in weather kit, seems to be pretty good

/// Magnus Formula
func calculateDewPoint(temp: Float, relativeHumidity: Float) -> Double {
	let a: Float = 17.27
	let b: Float = 237.7
	let alpha = ((a * temp) / (b + temp)) + log(relativeHumidity / 100.0)
	let dewPoint = (b * alpha) / (a - alpha)
	let dewPointUnit = Measurement<UnitTemperature>(value: Double(dewPoint), unit: .celsius)
	let locale = NSLocale.current as NSLocale
	let localeUnit = locale.object(forKey: NSLocale.Key(rawValue: "kCFLocaleTemperatureUnitKey"))
	var format: UnitTemperature = .celsius

	if localeUnit! as? String == "Fahrenheit" {
		format = .fahrenheit
	}
	return dewPointUnit.converted(to: format).value
}