meshtastic/Meshtastic-Apple #318: Set manually chance encryption key

Hraph, 2023-02-19:
### OS

iOS, iPadOS, macOS

### Description

Hello,

Why can't we manually set a channel encryption key in iOS app?
If I want to use the same key amongst all of my nodes, am I obliged to use the QR code sharing feature?

Thx

garthvh, 2023-02-19:
There would be a bunch of extra validation and it is less secure. The QR code or shared link works well.

gazreese, 2024-02-19:
Can we re-open this please @garthvh . This used to work (though was buggy due to I assume some poor validation code), and there's no way to create a channel from a PSK.

I also can't find a way to delete a channel.

People share the PSK strings all of the time and they're no less secure than putting a QR code somewhere. Can we please re-open the bug and fix the validation instead?

thebentern, 2024-02-19:
> This used to work (though was buggy due to I assume some poor validation code), and there's no way to create a channel from a PSK.

TestFlight has a new version of this with fixes, so it will likely be in the next App Store release.