meshtastic/Meshtastic-Apple #330: [Feature Request]: Add ability to share only secondary channels

thebentern, 2023-03-09:
### OS

iOS

### Description

I would like to be able to deselect primary channel when sharing QR codes

garthvh, 2023-03-09:
Based on trying this when I built the code generator I don't think this works / is really confusing for most users.

thebentern, 2023-03-09:
> Based on trying this when I built the code generator I don't think this works / is really confusing for most users. The webui offered this for a while and it did not work right.
This is probably due to channel indexing issues. I would propose that if you're only sharing a secondary channel, you're typically adding a "friend" to the mesh to setup their nodes, and in that case you'd want to share that as a single primary channel.

garthvh, 2023-03-24:
Will release with 2.1.3

closed 2023-03-24 by garthvh