meshtastic/Meshtastic-Apple #362: [Feature Request]: Visually not clear where to click to save

TheRealCryss, 2023-05-21:
### OS

macOS

### Description

It isn't obvious where exactly to click in order to save the settings. There should be a visual hint for that when the mouse is not over the area and when it hovers over it as well. Clicking on Save (Speichern in DE) which highlights up when there was a change in the input fields doesn't save, which one would expect. Instead you have to click in a part of the pop-up message.

This report is for the current Meshtastic app from the Appstore on MacOS. I haven't tested it on other OSs.

[image]

garthvh, 2023-05-21:
This is a Mac Catalyst issue, hopefully they enhance this control soon

garthvh, 2023-05-21:
Duplicate of https://github.com/meshtastic/Meshtastic-Apple/issues/237

closed 2023-05-21 by garthvh