meshtastic/Meshtastic-Apple #369: NRF52 Battery voltage reading is unreliable

charminULTRA, 2023-07-25:
### OS

iOS

### Description

Note to self for investigation - Will provide detail when I have the right data and evidence.

caveman99, 2023-07-25:
Observation is correct, no easy fix. nrf52 samples the battery voltage with an ADC. Several times. And calculates an average. Still the readings seem to fluctuate with TX activity. It's just as much as we can do currently short of adding a AXP type chip to the designs.

garthvh, 2023-07-25:
[image]
[image]

What battery are you using? I get really consistent readings with adafruit batteries.

charminULTRA, 2023-07-26:
First, voltage during charging appears over 4.20 volts, which isn't safe as far as I'm aware.

This is LINK on a RAK 19003.[image]