meshtastic/Meshtastic-Apple #381: Flickering signal graphic

mc-hamster, 2023-08-23:
### Firmware Version

Latest

### What did you do?

On the Bluetooth screen, the signal graphic is flickering. May be a symptom of data coming on the edge of two display values. May need a hysteresis algorithm to stabilize. 

### Expected Behavior

Stable rf indicator. 

### Participation

- [ ] I am willing to submit a pull request for this issue.

mc-hamster, 2023-08-23:
https://github.com/meshtastic/Meshtastic-Apple/assets/12804989/2e4511b7-4d84-434f-b34b-fb352a72dcbb

garthvh, 2023-08-23:
This is expected, those values update frequently, nrf connect (where the indicator code is from) does the same.

closed 2023-09-16 by garthvh