meshtastic/Meshtastic-Apple #383: 🚀 [Feature Request]: Localisation Support

kokroo, 2023-08-24:
### OS

iOS, iPadOS, macOS

### Description

As far as I can see, there is no support for localization in the Apple ecosystem apps yet.
I believe it's important to add support for more languages to make it easier for our friends around the world to adopt and use Meshtastic.

I am willing to submit the translations for Hindi to begin with, through Crowdin, preferably.

### Participation

- [ ] I am willing to submit a pull request for this issue.

closed 2023-08-24 by garthvh

geeksville, 2023-08-24:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/localisation-support-for-ios-macos/8069/1

kokroo, 2023-08-24:
> There are localized files for both German and Chinese already.

My bad then, but is this project linked to Crowdin in any way? Also, how can we help you add more languages?