meshtastic/Meshtastic-Apple #385: 🚀 [Feature Request]:  Air Quality Telemetry

garthvh, 2023-08-27:
### OS

iOS, iPadOS, macOS

### Description

* [ ] Air Quality Log
* [ ] AQI Chart / multicolor icon
* [ ] Calculate AQU once we have 50 samples

### Participation

- [ ] I am willing to submit a pull request for this issue.

Lolagatorade, 2024-12-24:
Not sure if it needs a new topic but regarding the newest updates and the great new graphs small issue related to the air quality is that the setting does not stick

Let's say I select to show air quality as soon as I leave the page once I go back in the setting has forgotten it should display it on the graph. 

[image]
[image]

jake-b, 2024-12-24:
Currently the graph and table settings do not persist.  This is a good feature request, and I’m hoping to eventually implement it.