meshtastic/Meshtastic-Apple #392: 🐞 [Bug]: Nodes showing up multiple times.

km6rpt, 2023-09-04:
### Firmware Version

2.1.23

### What did you do?

I apologize if I am doing something wrong.  I am a new use.  I have two Nano G1 Explorers.
They are both running the latest beta version 2.1.23
I have an Iphone 13 Pro running IOS 16.6.

Under the nodes section of the app, the information appears to be incorrect.
First, it shows the exact same node connected twice: Meshtastic 3838
Second, it is incorrectly showing Meshtastic 383c still connected to power.  This may just be an updating delay issue and not a bug.
I turned off Meshtastic 383c and now I see three nodes reporting for Meshtastic 383c instead of just one.

See attached screen shots.

If you would like for me to test with an alpha version or any other features, please let me know.
I can be reached at sdtoddl @ gmail

### Expected Behavior

Nodes showing up only once. 
[image]
[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

km6rpt, 2023-09-05:
I decided to update to the latest alpha version and so far I have been unable to replicate the bug above.

garthvh, 2023-09-06:
This has been released to the App Store