meshtastic/Meshtastic-Apple #393: 🚀 [Feature Request]:

mchuzie, 2023-09-05:
### OS

iPadOS

### Description

Request import/export config ability like android version

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2023-09-05:
I don't really see the need for administrative features in the mobile app so I am not likely to pick this up, if someone wants to submit code that creates and imports the same .cfg file they can.

garthvh, 2023-09-07:
What settings that are not part of the QR code need configuration on every device?

mchuzie, 2023-09-07:
Not everyone has a license and not every device belongs to the same group/channel all the time when running multiple groups with different encryption. Also, some groups of devices are better suited for different power consumption, location settings, etc, so the ability to import pre-canned config files would make setup much easier. Lastly, why have a 2 mobile platforms that aren’t offering the same functionality?

garthvh, 2023-09-11:
This all seems like administrative functionality that probably does not need to be part of the mobile app.  These are different platforms with different capabilities and developers.  I don't have any of those power settings on iOS for the same reason, they are used wrong more than correctly and are necessary only for a few edge cases.