meshtastic/Meshtastic-Apple #398: 🚀 [Feature Request]: MapKit for SwiftUI Updates

garthvh, 2023-09-15:
### OS

iOS, iPadOS, macOS

### Description

Update Maps to use MapKit for SwiftUI 

### Participation

- [X] I am willing to submit a pull request for this issue.

wschnell, 2023-10-02:
Hi,
looks like MapKit for SwiftUI is not in the package. Maybe needs to be bundled.
I installed the Meshtastic App from the Apple Store on my iMac running macOS 13.5.2 and the app crashes on start.
I do not have XCode or other developer components installed on this Mac.
thanks
Walter

garthvh, 2023-10-02:
> Hi, looks like MapKit for SwiftUI is not in the package. Maybe needs to be bundled. I installed the Meshtastic App from the Apple Store on my iMac running macOS 13.5.2 and the app crashes on start. I do not have XCode or other developer components installed on this Mac. thanks Walter

Yeah it is only available in iOS 17 and macOS 14, not really sure why Intel Macs are trying to load it

closed 2023-11-20 by garthvh