meshtastic/Meshtastic-Apple #400: 🐞 [Bug]: 100% hard crash macos 13.6 Intel

rinchen, 2023-09-21:
### Firmware Version

2.1.23

### What did you do?

Latest app store version 2.2.6 installed. Launch app and it crashes with abend report.

### Expected Behavior

App doesn't crash

### Current Behavior

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Meshtastic [13953]
Path:                  /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
Identifier:            gvh.MeshtasticClient
Version:               2.2.6 (667)
App Item ID:           1586432531
App External ID:       859960242
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-09-21 14:32:30.4338 -0600
OS Version:            macOS 13.6 (22G120)
Report Version:        12
Bridge OS Version:     8.0 (21P365)
Anonymous UUID:        2CE179D2-EE4C-D7E2-EDAB-1C1F2AC8C41D

Time Awake Since Boot: 4500 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 4 Symbol missing
Symbol not found: _$s15_MapKit_SwiftUI0A11PitchSliderV0cD04ViewAAMc
Referenced from: <F11861AB-7A70-3A0F-AE89-C7D7C87616DF> /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
Expected in:     <CB344F83-E857-3480-9884-113B7945A3AD> /System/iOSSupport/System/Library/Frameworks/_MapKit_SwiftUI.framework/Versions/A/_MapKit_SwiftUI
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	    0x7ff816ce9c52 __abort_with_payload + 10
1   dyld                          	    0x7ff816d03fd7 abort_with_payload_wrapper_internal + 82
2   dyld                          	    0x7ff816d04009 abort_with_payload + 9
3   dyld                          	    0x7ff816c888f0 dyld4::halt(char const*) + 375
4   dyld                          	    0x7ff816c85b71 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 4526
5   dyld                          	    0x7ff816c843bd start + 1805

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7b2a82b88  rdx: 0x00007ff7b2a82ff0
  rdi: 0x0000000000000006  rsi: 0x0000000000000004  rbp: 0x00007ff7b2a82bd0  rsp: 0x00007ff7b2a82b88
   r8: 0x00007ff7b2a82bf0   r9: 0x0000000000000000  r10: 0x00000000000000df  r11: 0x0000000000000246
  r12: 0x00000000000000df  r13: 0x00007ff7b2a82ff0  r14: 0x0000000000000004  r15: 0x0000000000000006
  rip: 0x00007ff816ce9c52  rfl: 0x0000000000000246  cr2: 0x000000010db4e000
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
       0x10d479000 -        0x10d9e8fff gvh.MeshtasticClient (2.2.6) <f11861ab-7a70-3a0f-ae89-c7d7c87616df> /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
[... 175 lines cut ...]
USB Device: CalDigit Thunderbolt 3 Audio
USB Device: USB30Bus
USB Device: Card Reader
USB Device: C922 Pro Stream Webcam
USB Device: USB31Bus
USB Device: USB2.0 Hub
USB Device: Das Keyboard
USB Device: Mobile Drive
USB Device: USB3.0 Hub
USB Device: T2Bus
USB Device: composite_device
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Device: TS3 Plus, CalDigit, Inc., 3, 44.1
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5

### Participation

- [ ] I am willing to submit a pull request for this issue.

rinchen, 2023-09-21:
I tested on an m2 mac and got the same results:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Meshtastic [10448]
Path:                  /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
Identifier:            Meshtastic
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-09-21 14:41:28.0251 -0600
OS Version:            macOS 13.6 (22G120)
Report Version:        12
Anonymous UUID:        346C7370-EC84-8B5A-4245-FF05D9F2BB4F

Time Awake Since Boot: 5600 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 4 Symbol missing
Symbol not found: _$s15_MapKit_SwiftUI0A11PitchSliderV0cD04ViewAAMc
Referenced from: <35997646-0A0D-303B-B254-030106EC478B> /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
Expected in:     <E80FC237-FE53-33AA-9D96-9BD89489CEC1> /System/iOSSupport/System/Library/Frameworks/_MapKit_SwiftUI.framework/Versions/A/_MapKit_SwiftUI
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	       0x18435d118 __abort_with_payload + 8
1   dyld                          	       0x184368d7c abort_with_payload_wrapper_internal + 104
2   dyld                          	       0x184368db0 abort_with_payload + 16
3   dyld                          	       0x1842f48a8 dyld4::halt(char const*) + 328
4   dyld                          	       0x1842f1590 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 4184
5   dyld                          	       0x1842efe18 start + 1964

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000006   x1: 0x0000000000000004   x2: 0x000000016eea0010   x3: 0x00000000000000df
    x4: 0x000000016ee9fc10   x5: 0x0000000000000000   x6: 0x0000000000000000   x7: 0x000000016ee9f5f8
    x8: 0x0000000000000020   x9: 0x0000000000000009  x10: 0x0000000000000001  x11: 0x000000000000000a
   x12: 0x0000000000000000  x13: 0x0000000000000031  x14: 0x000000023239e650  x15: 0x0000000000008000
   x16: 0x0000000000000209  x17: 0x00000001842ed35c  x18: 0x0000000000000000  x19: 0x0000000000000000
   x20: 0x000000016ee9fc10  x21: 0x00000000000000df  x22: 0x000000016eea0010  x23: 0x0000000000000004
   x24: 0x0000000000000006  x25: 0x0000000000001480  x26: 0x0000000000000001  x27: 0x000000001f070004
   x28: 0x00000001015e6460   fp: 0x000000016ee9fbe0   lr: 0x0000000184368d7c
    sp: 0x000000016ee9fba0   pc: 0x000000018435d118 cpsr: 0x00001000
   far: 0x00000001015fc000  esr: 0x56000080  Address size fault

Binary Images:
       0x100f5c000 -        0x10148ffff Meshtastic (*) <35997646-0a0d-303b-b254-030106ec478b> /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
       0x1842ea000 -        0x184378587 dyld (*) <49204446-242b-3d1e-9704-32f8ac99723e> /usr/lib/dyld

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by all processes on this machine:
    task_for_pid: 0
    thread_create: 0
[... 110 lines cut ...]
      },
      "deploymentId" : 240000027
    },
    {
      "rolloutId" : "60186475825c62000ccf5450",
      "factorPackIds" : {

      },
      "deploymentId" : 240000062
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "3a3cf641-8471-4e4e-9ad4-81d0ede970fd",
      "experimentId" : "64a84dae90d82611a0bd7d3d",
      "deploymentId" : 400000004
    },
    {
      "treatmentId" : "6dd670af-0633-45e4-ae5f-122ae4df02be",
      "experimentId" : "64406ba83deb637ac8a04419",
      "deploymentId" : 900000017
    }
  ]
}
}

garthvh, 2023-09-21:
When did 13.6 release? I have never seen it in the logs.

rinchen, 2023-09-21:
Today. It also abended the same way on 13.5 for me.  iOS app works fine fwiw.

garthvh, 2023-09-26:
Can you try the latest Mac app store release? 2.2.7

jlrgraham, 2023-09-26:
For me 2.2.7 is still crashing on macOS 13.5.2:
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Meshtastic [74206]
Path:                  /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
Identifier:            gvh.MeshtasticClient
Version:               2.2.7 (673)
App Item ID:           1586432531
App External ID:       860197345
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-09-25 20:56:30.9042 -0500
OS Version:            macOS 13.5.2 (22G91)
Report Version:        12
Bridge OS Version:     7.6 (20P6072)
Anonymous UUID:        EE6F7A94-850A-79BF-D15C-66F11472D760

Time Awake Since Boot: 270000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 4 Symbol missing
Symbol not found: _$s15_MapKit_SwiftUI0A11PitchSliderV0cD04ViewAAMc
Referenced from: <0EE1A217-F85A-39F7-9998-96128FA39805> /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
Expected in:     <CB344F83-E857-3480-9884-113B7945A3AD> /System/iOSSupport/System/Library/Frameworks/_MapKit_SwiftUI.framework/Versions/A/_MapKit_SwiftUI
(terminated at launch; ignore backtrace)

...
```

rinchen, 2023-09-26:
2.2.7 on 13.6 abends.

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Meshtastic [44304]
Path:                  /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
Identifier:            gvh.MeshtasticClient
Version:               2.2.7 (673)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-09-25 21:22:47.5428 -0600
OS Version:            macOS 13.6 (22G120)
Report Version:        12
Bridge OS Version:     8.0 (21P365)
Anonymous UUID:        2CE179D2-EE4C-D7E2-EDAB-1C1F2AC8C41D

Sleep/Wake UUID:       A6EEAB5D-CC8C-4625-BFE3-EDD16B0B8646

Time Awake Since Boot: 330000 seconds
Time Since Wake:       96232 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 4 Symbol missing
Symbol not found: _$s15_MapKit_SwiftUI0A11PitchSliderV0cD04ViewAAMc
Referenced from: <0EE1A217-F85A-39F7-9998-96128FA39805> /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
Expected in:     <CB344F83-E857-3480-9884-113B7945A3AD> /System/iOSSupport/System/Library/Frameworks/_MapKit_SwiftUI.framework/Versions/A/_MapKit_SwiftUI
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	    0x7ff816ce9c52 __abort_with_payload + 10
1   dyld                          	    0x7ff816d03fd7 abort_with_payload_wrapper_internal + 82
2   dyld                          	    0x7ff816d04009 abort_with_payload + 9
3   dyld                          	    0x7ff816c888f0 dyld4::halt(char const*) + 375
4   dyld                          	    0x7ff816c85b71 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 4526
5   dyld                          	    0x7ff816c843bd start + 1805

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7b6adeb88  rdx: 0x00007ff7b6adeff0
  rdi: 0x0000000000000006  rsi: 0x0000000000000004  rbp: 0x00007ff7b6adebd0  rsp: 0x00007ff7b6adeb88
   r8: 0x00007ff7b6adebf0   r9: 0x0000000000000000  r10: 0x00000000000000df  r11: 0x0000000000000246
  r12: 0x00000000000000df  r13: 0x00007ff7b6adeff0  r14: 0x0000000000000004  r15: 0x0000000000000006
  rip: 0x00007ff816ce9c52  rfl: 0x0000000000000246  cr2: 0x0000000109b06000
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
       0x10941d000 -        0x109998fff gvh.MeshtasticClient (2.2.7) <0ee1a217-f85a-39f7-9998-96128fa39805> /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
    0x7ff816c7e000 -     0x7ff816d165ef dyld (*) <3df96f32-b9c9-3566-a6b7-4daebc6d6563> /usr/lib/dyld

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
[... 162 lines cut ...]
USB Device: USB31Bus
USB Device: hub_device
USB Device: USB Controls
USB Device: USB30Bus
USB Device: CalDigit Thunderbolt 3 Audio
USB Device: USB30Bus
USB Device: Card Reader
USB Device: C922 Pro Stream Webcam
USB Device: USB31Bus
USB Device: USB2.0 Hub
USB Device: Das Keyboard
USB Device: Mobile Drive
USB Device: USB3.0 Hub
USB Device: T2Bus
USB Device: composite_device
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5
Thunderbolt Device: TS3 Plus, CalDigit, Inc., 3, 44.1
Thunderbolt Bus: MacBook Pro, Apple Inc., 63.5

rinchen, 2023-10-07:
2.2.7 is working on Sonoma 14.0

jlrgraham, 2023-10-11:
Greetings, unfortunately the app still crashes for me on startup with `2.2.8`:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Meshtastic [49451]
Path:                  /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
Identifier:            gvh.MeshtasticClient
Version:               2.2.8 (686)
App Item ID:           1586432531
App External ID:       860447668
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-10-10 22:03:26.0673 -0500
OS Version:            macOS 13.5.2 (22G91)
Report Version:        12
Bridge OS Version:     7.6 (20P6072)
Anonymous UUID:        EE6F7A94-850A-79BF-D15C-66F11472D760

Time Awake Since Boot: 1500000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 4 Symbol missing
Symbol not found: _$s15_MapKit_SwiftUI0A11PitchSliderV0cD04ViewAAMc
Referenced from: <0EBEA9F3-85EE-3A7F-BBD4-55C81CAEA37E> /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
Expected in:     <CB344F83-E857-3480-9884-113B7945A3AD> /System/iOSSupport/System/Library/Frameworks/_MapKit_SwiftUI.framework/Versions/A/_MapKit_SwiftUI
(terminated at launch; ignore backtrace)

Thread 0 Crashed:
0   dyld                          	    0x7ff8008c5c52 __abort_with_payload + 10
1   dyld                          	    0x7ff8008dffd7 abort_with_payload_wrapper_internal + 82
2   dyld                          	    0x7ff8008e0009 abort_with_payload + 9
3   dyld                          	    0x7ff8008648f0 dyld4::halt(char const*) + 375
4   dyld                          	    0x7ff800861b71 dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 4526
5   dyld                          	    0x7ff8008603bd start + 1805

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7b767fb68  rdx: 0x00007ff7b767ffd0
  rdi: 0x0000000000000006  rsi: 0x0000000000000004  rbp: 0x00007ff7b767fbb0  rsp: 0x00007ff7b767fb68
   r8: 0x00007ff7b767fbd0   r9: 0x0000000000000000  r10: 0x00000000000000df  r11: 0x0000000000000246
  r12: 0x00000000000000df  r13: 0x00007ff7b767ffd0  r14: 0x0000000000000004  r15: 0x0000000000000006
  rip: 0x00007ff8008c5c52  rfl: 0x0000000000000246  cr2: 0x0000000108f6d000
  
Logical CPU:     0
Error Code:      0x02000209 
Trap Number:     133

Binary Images:
       0x10887c000 -        0x108dfffff gvh.MeshtasticClient (2.2.8) <0ebea9f3-85ee-3a7f-bbd4-55c81caea37e> /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
    0x7ff80085a000 -     0x7ff8008f25ef dyld (*) <31e1c182-e611-3883-97a3-4334a21c90bd> /usr/lib/dyld

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
[... 138 lines cut ...]
Model: Macmini8,1, BootROM 1968.140.7.0.0 (iBridge: 20.16.6072.0.0,0), 6 processors, 6-Core Intel Core i7, 3.2 GHz, 64 GB, SMC 
Graphics: Intel UHD Graphics 630, Intel UHD Graphics 630, Built-In
Display: LG HDR 4K, 3840 x 2160 (2160p/4K UHD 1 - Ultra High Definition), Main, MirrorOff, Online
Memory Module: BANK 0/ChannelA-DIMM0, 32 GB, DDR4, 2667 MHz, 859B, CT32G4SFD8266.C16FB
Memory Module: BANK 2/ChannelB-DIMM0, 32 GB, DDR4, 2667 MHz, 859B, CT32G4SFD8266.C16FB
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: Dec  9 2022 16:56:24 version 9.30.492.0.32.5.87 FWID 01-3a11ec19
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: LAN100, Ethernet, vlan1
Network Service: Wi-Fi, AirPort, en1
Network Service: Thunderbolt Bridge, Ethernet, bridge0
USB Device: USB31Bus
USB Device: CP2104 USB to UART Bridge Controller
USB Device: hub_device
USB Device: USB2.0 Hub
USB Device: Das Keyboard
USB Device: YubiKey OTP+FIDO+CCID
USB Device: USB Single Serial
USB Device: USB Controls
USB Device: WisCore RAK4631 Board
USB Device: T2Bus
USB Device: Headset
USB Device: Apple T2 Controller
Thunderbolt Bus: Mac mini, Apple Inc., 47.5
Thunderbolt Bus: Mac mini, Apple Inc., 47.5
```

closed 2023-11-20 by garthvh