meshtastic/Meshtastic-Apple #402: 🚀 [Feature Request]: nodes on the Brazil map

Marlontebaldi, 2023-09-27:
### OS

iOS

### Description

Why don't the nodes appear on the Brazil map when I'm on the iOS app?

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2023-09-27:
What is the Brazil map?

Marlontebaldi, 2023-09-27:
> What is the Brazil map?

When I create a fake location in any country other than Brazil, the node appears on the app's map, but when I put the location in Brazil, the app doesn't show it

Marlontebaldi, 2023-09-27:
> What is the Brazil map?

nodes located in Brazil don't appear on the map. If I fake a location - anywhere other than Brazil, the app displays it correctly

garthvh, 2023-09-27:
There is nothing preventing waypoints or nodes in Brazil, just added a waypoint and set a fixed position, I would review the Position setup on your node.

closed 2023-09-27 by garthvh

Marlontebaldi, 2023-09-27:
[image]
[image]
[image]
[image]
[image]
[image]

on android it works normally

> There is nothing preventing waypoints or nodes in Brazil, just added a waypoint and set a fixed position, I would review the Position setup on your node.

Marlontebaldi, 2023-09-27:
> There is nothing preventing waypoints or nodes in Brazil, just added a waypoint and set a fixed position, I would review the Position setup on your node.

waypoints seem to work ok. the problem is with device positions. I have reproduced the problem by using a fixed position on my device (as show in the serial commands below)

roger-inspiron:~$ meshtastic --setlat 20 --setlon 20
Connected to radio
Fixing latitude at 20.0 degrees
Fixing longitude at 20.0 degrees
Setting device position
roger-inspiron:~$ meshtastic --setlat -20 --setlon 20
Connected to radio
Fixing latitude at -20.0 degrees
Fixing longitude at 20.0 degrees
Setting device position
roger-inspiron:~$ meshtastic --setlat -20 --setlon -20
Connected to radio
Fixing latitude at -20.0 degrees
Fixing longitude at -20.0 degrees
Setting device position
roger-inspiron:~$ meshtastic --setlat 20 --setlon -20
Connected to radio
Fixing latitude at 20.0 degrees
Fixing longitude at -20.0 degrees
Setting device position

[image]

garthvh, 2023-10-26:
Try with the latest version of the app, I am seeing quite a few southern positions on the public MQTT

Marlontebaldi, 2023-10-26:
> Try with the latest version of the app, I am seeing quite a few southern positions on the public MQTT
[image]

Now it's working, thank you, do you have any idea of ​​putting these iOS features in the Android app?

garthvh, 2023-10-27:
These are all features of apple's maps, android is a whole different set of maps.