meshtastic/Meshtastic-Apple #406: 🚀 [Feature Request]: Show routers on the Mesh Map

wschnell, 2023-10-08:
### OS

iOS, iPadOS, macOS

### Description

current behavior: 
client nodes are shown on the mesh map

expected behavior:
client nodes AND routers are shown on the map
---------------
greetings and keep up the good work!
bye
Walter

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2023-10-08:
There is nothing in the app restricting routers from showing, repeaters don't send any positions so they won't be on the mesh map.

wschnell, 2023-10-08:
Thanks Garth,
You are right, we have a convention for naming nodes here around Salzburg (begin the name with "R_"), so I cannot see if it is a router or a repeater, and 3 out of 4 do not send position data. The one that sends is on the map.
case closed
sorry for taking your time.
The Apple App is really nice by now.
bye
Walter

closed 2023-10-09 by garthvh