meshtastic/Meshtastic-Apple #408: 🐞 [Bug]: switching between channels does not work in UI

wschnell, 2023-10-08:
### Firmware Version

2.2.10

### What did you do?

I have the Primary channel as 0 AND a secondary channel as 1.
In the Apple Client, under messages, I want to switch between the 2 channels.

This happens in iOS (newest version) on my iPad AND in macOS 14.0 on my iMac.
Apple Client is 2.2.7 on each.

### Expected Behavior

the leftmost window shows the messages of the chosen channel.

### Current Behavior

the leftmost window does not switch between the (sets of) messages of the chosen channel.

### Participation

- [ ] I am willing to submit a pull request for this issue.

wschnell, 2023-10-23:
same program logic as with switching between the DM threads, which works OK.

wschnell, 2023-11-06:
workaround: close the Apple Meshtastic app and reopen, go to messages and choose the channel you want.

garthvh, 2023-12-21:
Fixed in https://github.com/meshtastic/Meshtastic-Apple/pull/434

closed 2023-12-21 by garthvh