meshtastic/Meshtastic-Apple #409: 🚀 [Feature Request]: play sound when new message is received

wschnell, 2023-10-10:
### OS

iOS, iPadOS, macOS

### Description

The App could play a sound when a new message is received. On-Off in Settings. 
The luxury version would make the sound file configurable in Settings.

Side note:
There is the idea of connecting a buzzer or a speaker to the node device - this is completely unnecessary when it is connected to a Mac or iPad or iPhone.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2023-10-11:
[image]

This is managed by iOS

closed 2023-10-11 by garthvh

wschnell, 2023-10-13:
Hi,
on my iPad, all notification options were set to ON upon installation, I get popup notifications, but no sound.
on my iMac, all notification were set to OFF upon installation, I set them to ON => I get popups, but no sound.
I will report if I find that I am doing something wrong ...

garthvh, 2023-10-14:
On all three platforms permission on first launch pop ups are presented for BLE, location and notification permissions.