meshtastic/Meshtastic-Apple #412: 🚀 [Feature Request]: MQTT Only Virtual Node

garthvh, 2023-10-11:
### OS

iOS, iPadOS, macOS

### Description

Build out the necessary structures to create a virtual MQTT only node that does not require a lora radio.

*  [ ] Generate a unique node from the phone
*  [ ] Subscribe to MQTT 
*  [ ] Acknowledge Message
*  [ ] Disable most device and module settings

### Participation

- [X] I am willing to submit a pull request for this issue.

AssetBurned, 2024-02-11:
If you create a virtual node for MQTT, then it might be possible to also extend that to the network. Kind of having a whole virtual node.

Reason for that would be to have a portable node. With the [continuity](https://www.apple.com/macos/continuity/) options, it then would be possible to start a conversation e.g. inside the iOS app and continue on the Mac.

garthvh, 2024-06-08:
Good explanation of the complications with this here https://github.com/meshtastic/Meshtastic-Android/issues/361 can revisit if the firmware adds support.

closed 2024-06-08 by garthvh