meshtastic/Meshtastic-Apple #417: 🐞 [Bug]: Settings area is unusable after switching devices

bmidgley, 2023-10-20:
### Firmware Version

2.1.23

### What did you do?

Pair with two meshtastic devices on bluetooth. Connect to device 1. Open settings, make a change, and save it. Disconnect and connect to device 2. 

### Expected Behavior

The settings area should be immediately usable after switching devices.

### Current Behavior

There are two possible scenarios which are both unusable.

* Data from device 1 is still showing in the settings area
* Settings are locked and a message says it is waiting for a response from the device

The workaround is to force quit the application if you ever have to switch between devices.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

I have a fixed pin in case that matters. Happens with any meshtastic devices, not vendor specific.

garthvh, 2023-10-21:
There is no lingering data from the last device when you change to a new device, it deletes every thing in the app including app settings and the preferred peripheral. If you go to the settings screen before the bluetooth connection is done connecting the settings will be locked, but will become available when the connection completes.

bmidgley, 2023-10-25:
https://github.com/meshtastic/Meshtastic-Apple/assets/63477/6960b5b6-c909-4e50-a89b-e11f5adfb146

The app is still "configuring" radcom1 after switching to decked1. So maybe technically it is showing you the right values for the device it's configuring, but it's confusing to have to change the device I'm configuring after I've already changed the device I'm connected to. And after a bit, it seems to change on its own to configuring decked1. I don't think I changed that.

bmidgley, 2023-10-25:
https://github.com/meshtastic/Meshtastic-Apple/assets/63477/77d2fa02-d244-44d2-a7ea-b3176f639605

In this one, after changing from decked1 to radcom1, I look at the name and it's blank. I don't know what I'm configuring. I go back and now I'm configuring decked1. Then I go back again and I'm configuring radcom1. I can see now it's the device being configured that is changing and unpredictable and leads to confusion about whose settings I'm looking at or changing.

garthvh, 2023-10-26:
So while the orange communicating with device animation is running on the bluetooth screen all of the config packets are being imported and saved.  When you click straight over those settings can still be empty.  If you are on mqtt or have a big network the connection is getting kind of long, we have been looking at some options to separate things a bit more.

closed 2023-12-21 by garthvh