meshtastic/Meshtastic-Apple #423: 🚀 [Feature Request]: FaceID/Password lock on the app for safety

RCGV1, 2023-11-05:
### OS

iOS, iPadOS

### Description

I think it would be useful for users who want added security for a lock option to be added to the app so the wrong people cannot message or read messages on their behalf.

### Participation

- [X] I am willing to submit a pull request for this issue.

garthvh, 2023-11-05:
I don't really want to manage an app only password, without some firmware support I am not sure what this would really do.

RCGV1, 2023-11-05:
There is a native way to do this. https://www.hackingwithswift.com/books/ios-swiftui/locking-our-ui-behind-face-id

garthvh, 2023-11-05:
Yeah I have looked at it before https://developer.apple.com/documentation/localauthentication/logging_a_user_into_your_app_with_face_id_or_touch_id/ it is meant to be used with some sort of backing login system, which the device does not have yet and does not exist in the app either.

garthvh, 2023-12-07:
This is only done by financial institutions, there is no Face ID lock for messages because it is too much friction.

closed 2023-12-07 by garthvh