meshtastic/Meshtastic-Apple #432: Mbtiles not loading🐞 [Bug]:

jbaize, 2023-12-08:
### Firmware Version

2.2.13

### What did you do?

Ipad with version 16.7.2.   Selected the options to enable offline maps and mbtiles.  Downloaded a mbtiles file (32Mb) and stored it locally on the device. Shared with Meshtastic.

### Expected Behavior

Expected that the mbtiles file would be loaded in Meshtastic.

### Current Behavior

Apple maps is shown instead of the mbtiles file when the offline map option is selected. 

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2023-12-08:
There is no way to turn off the Apple Maps and only use the mb tiles, the mb tiles file will be used if it is available at the zoom level you are at.  Have to have the legacy map option checked in app settings.

jbaize, 2023-12-08:
I did finally get this to work by "Clear App Data" and opening the map again.

closed 2023-12-08 by jbaize