meshtastic/Meshtastic-Apple #433: 🐞 [Bug]:

I3rixon, 2023-12-19:
### Firmware Version

2.2.14 last version

### What did you do?

Location issue.
Tried used fixed posotion. Set on Adnroid, then connect on iPhone - and app set it to zero/wiped.
Tried also from Python console version of Meshtastic - it worked, BUT calculation between nodes not correct. in reality it cople of km but in app it wrote ~6800 km like not found correct location
As I remember in prev version I don't have that issue.
I have tried reinstall and different versions of firmware - the same result.
Location from GPS in iPhone not set also, only fixed. Not shared location

### Expected Behavior

Correct location from iPhone GPS should be work

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2023-12-19:
When you enable the fixed position checkbox it sends your current location to the device as your position.

I3rixon, 2023-12-19:
no. it not worked. it set to zero lat/long/alt

I3rixon, 2023-12-19:
permision to GPS in granted for that app

garthvh, 2023-12-20:
Those positions are being sent into the phone, I don't see where you have set fixed position.

Are you on iOS 17?

I3rixon, 2023-12-20:
[image]
iOS 17.

I3rixon, 2023-12-20:
with fixed position as you can see it empty
[image]

garthvh, 2023-12-20:
I think you need to upgrade your firmware, that is probably the right distance to null island

closed 2023-12-20 by garthvh

I3rixon, 2023-12-20:
firmware is 2.2.16 is last one (tested other versions too). I have posted version of App what I used is 2.2.14 also last