meshtastic/Meshtastic-Apple #435: 🐞 [Bug]: Heltec v3 sets latlng to 0 when should be prexied from iOS device

geftactics, 2023-12-20:
### Firmware Version

2.2.15

### What did you do?

- Flashed Heltec device with 2.2.15
- Installed iOS app. Granted 'Always' for location services
- Validated iOS app was getting GPS data (Settings > App Settings > Phone GPS)
- Enabled 'Share location' (Settings > App Settings > Location Settings)
- Paired Heltec v3 via Bluetooth
- Wait a while
- Validated that Apple device is sending regular location data to node (Settings > Mesh Log)

### Expected Behavior

Expected the Heltec under this configuration to recieve GPS lat/lng from iOS, and then pass these onto the mesh. (Note, my node doesnt have a GPS module). 

I did also remove the app from my iOS device, reinstall and get the same results from another Heltec device

### Current Behavior

No location data appears to be passed. Using the CLI, I can see after waiting a good amount of time we just see N/A for lat/lng. I managed to extract the following debug log from the node, where we can see value of 0 for lat and lng...

```
DEBUG | 22:12:00 140 PACKET FROM PHONE (id=0x00000000 fr=0x94 to=0x94, WantAck=0, HopLim=0 Ch=0x0 Portnum=3)
WARN  | 22:12:00 140 phone tried to pick a nodenum, we don't allow that.
DEBUG | 22:12:00 140 Enqueued local (id=0x03e7079b fr=0x00 to=0x94, WantAck=0, HopLim=0 Ch=0x0 Portnum=3 rxtime=1703023920)
INFO  | 22:12:00 140 Telling client we have new packets 28
INFO  | 22:12:00 140 BLE notify fromNum
DEBUG | 22:12:00 140 [Router] Add packet record (id=0x03e7079b fr=0x00 to=0x94, WantAck=0, HopLim=0 Ch=0x0 Portnum=3 rxtime=1703023920)
DEBUG | 22:12:00 140 [Router] handleReceived(REMOTE) (id=0x03e7079b fr=0x00 to=0x94, WantAck=0, HopLim=0 Ch=0x0 Portnum=3 rxtime=1703023920)
DEBUG | 22:12:00 140 [Router] Module 'position' wantsPacket=1
INFO  | 22:12:00 140 [Router] Received position from=0x0, id=0x3e7079b, portnum=3, payloadlen=13
DEBUG | 22:12:00 140 [Router] Incoming update from MYSELF
DEBUG | 22:12:00 140 [Router] Setting local position: latitude=0, longitude=0, time=1703023791
INFO  | 22:12:00 140 [Router] POSITION node=e2e59094 l=13 latI=0 lonI=0 msl=0 hae=0 geo=0 pdop=0 hdop=0 vdop=0 siv=1 fxq=0 fxt=0 pts=1703023791 time=1703023791
```

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

I did also experiment with (Settings > Radio Config > Position > Device GPS) - Not quite sure what the config for my situation is... My node has no GPS and want to use GPS from the app - do I turn off 'Device GPS enabled'? If it do this, node firmware then states 'GPS disabled by button'

garthvh, 2023-12-20:
Did you set the GPS pins for the Heltec?  I can't replicate on iOS 17, are you guys on 16?  I would also recommend upgrading to the latest firmware.

geftactics, 2023-12-20:
iOS 17.1.2

I don’t have GPS module attached to Heltec node, so have left gpio config as per default. Will give the alpha version of firmware a try.

what settings did you set related to GPS when you managed to not replicate the issue?

garthvh, 2023-12-20:
I use the phone GPS exclusively, try turning the location settings on and off under settings, or enable the new always location option.

garthvh, 2023-12-21:
Default settings, should start sending locations based on your settings once share location is enabled under app settings.

You can also see the position the app is getting from the phone under app settings.

geftactics, 2023-12-21:
@garthvh - Is your testing with Heltec v3 board or some other non-gps node? I am wondering if issue is specific to the Heltec maybe?

I did confirm in my initial report (Step 3) that the app is 100% obtaining correct GPS data from iOS, as you mentioned, I can validate this is correct under settings > app settings > phone GPS, where i see my current latlng.

As advised, I have wiped my Heltec node and installed latest firmware `2.2.16.1c6acfd`, also re-installed the iOS app for good measure - the only setting changed is Settings > App Settings > Location Settings > share location = enabled. But I still see in the logs:
```
INFO  | 13:35:50 427 [Router] Received position from=0x0, id=0x63c185f, portnum=3, payloadlen=13
DEBUG | 13:35:50 427 [Router] Incoming update from MYSELF
DEBUG | 13:35:50 427 [Router] Setting local position: latitude=0, longitude=0, time=1703165381
INFO  | 13:35:50 427 [Router] POSITION node=e2e59350 l=13 latI=0 lonI=0 msl=0 hae=0 geo=0 pdop=0 hdop=0 vdop=0 siv=1 fxq=0 fxt=0 pts=1703165381 time=1703165381
```

Do you know what next stage of debugging is here? Can we see the payload the iOS is sending to the node over Bluetooth somehow? I can see that the iOS app is sending position packets from apple device GPS in the mesh log, but I cannot see the payload of that packet.

garthvh, 2023-12-22:
I am a little confused as to where it is getting the zeros from, I think it must be happening in the firmware.

I added some filtering to the p positions being output by the phone in the version just released to the App Store.

geftactics, 2023-12-22:
Just grabbed new version from app store and installed - App is no longer logging Apple device positioning messages in the mesh log (But think that is another change in this version)... Logs are still saying 0 for latlng...
```
DEBUG | 08:13:03 673 [Router] Add packet record (id=0x3492c03d fr=0x00 to=0x50, WantAck=0, HopLim=0 Ch=0x0 Portnum=3 rxtime=1703232783)
DEBUG | 08:13:03 673 [Router] handleReceived(REMOTE) (id=0x3492c03d fr=0x00 to=0x50, WantAck=0, HopLim=0 Ch=0x0 Portnum=3 rxtime=1703232783)
DEBUG | 08:13:03 673 [Router] Module 'position' wantsPacket=1
INFO  | 08:13:03 673 [Router] Received position from=0x0, id=0x3492c03d, portnum=3, payloadlen=13
DEBUG | 08:13:03 673 [Router] Incoming update from MYSELF
DEBUG | 08:13:03 673 [Router] Setting local position: latitude=0, longitude=0, time=1703235183
INFO  | 08:13:03 673 [Router] From Radio onread
INFO  | 08:13:03 673 [Router] POSITION node=e2e59350 l=13 latI=0 lonI=0 msl=0 hae=0 geo=0 pdop=0 hdop=0 vdop=0 siv=1 fxq=0 fxt=0 pts=1703235183 time=1703235183
INFO  | 08:13:03 673 [Router] getFromRadio=STATE_SEND_PACKETS
INFO  | 08:13:03 673 [Router] updatePosition REMOTE node=0xe2e59350 time=1703235183, latI=0, lonI=0
```

Would it be an idea to retain the Apple GPS position rows in the mush log, and also include the latlng that was sent? or even the ability to click on a log row in meshlog to reveal the exact payload that was sent, to assist with debugging?

Is there some extra debugging that I can do on the firmware side to try and work out what's going on?

I3rixon, 2023-12-22:
@garthvh firmware is ok, if set from android app or python client - position worked ok, but from iPhone app it not worked at all.

geftactics, 2023-12-22:
@I3rixon - Are you able to share the same firmware logs that are generated from an Android device doing the update, so that we can compare? I don't have an android device, so can't do this!

I3rixon, 2023-12-22:
**iPhone:**
```
INFO  | 11:26:36 23 [Router] From Radio onread
DEBUG | 11:26:36 23 [Router] Incoming update from MYSELF
DEBUG | 11:26:36 23 [Router] Setting local position: latitude=0, longitude=0, time=1703244380
INFO  | 11:26:36 23 [Router] POSITION node=1f37e46c l=13 latI=0 lonI=0 msl=0 hae=0 geo=0 pdop=0 hdop=0 vdop=0 siv=1 fxq=0 fxt=0 pts=1703244380 time=1703244380
```

**Android:**
```
DEBUG | 11:29:09 176 [Router] Incoming update from MYSELF
DEBUG | 11:29:09 176 [Router] Setting local position: latitude=478892461, longitude=333090800, time=1703250642
INFO  | 11:29:09 176 [Router] POSITION node=1f37e46c l=17 latI=478892461 lonI=333090800 msl=87 hae=0 geo=0 pdop=0 hdop=0 vdop=0 siv=0 fxq=0 fxt=0 pts=0 time=1703250642
INFO  | 11:29:09 176 [Router] updatePosition REMOTE node=0x1f37e46c time=1703250642, latI=478892461, lonI=333090800
DEBUG | 11:29:09 176 [Router] Node status update: 5 online, 5 total
```

garthvh, 2023-12-22:
> **iPhone:**
>
> ```
> INFO  | 11:26:36 23 [Router] From Radio onread
> DEBUG | 11:26:36 23 [Router] Incoming update from MYSELF
> DEBUG | 11:26:36 23 [Router] Setting local position: latitude=0, longitude=0, [...]

What firmware version is this?

I3rixon, 2023-12-22:
Last version 2.2.16 also tested on old versions the same

I3rixon, 2023-12-22:
also strange here on iPhone:
INFO  | 11:27:12 60 To Radio onwrite
DEBUG | 11:27:12 60 **PACKET FROM PHONE** (id=**0x00000000** fr=0x6c to=0x6c, WantAck=0, HopLim=0 Ch=0x0 Portnum=3)
WARN  | 11:27:12 60 **phone tried to pick a nodenum, we don't allow that.**
DEBUG | 11:27:12 60 Enqueued local (id=0x4ee8f68d fr=0x00 to=0x6c, WantAck=0, HopLim=0 Ch=0x0 Portnum=3 rxtime=1703244432)
DEBUG | 11:27:12 60 [Router] Add packet record (id=0x4ee8f68d fr=0x00 to=0x6c, WantAck=0, HopLim=0 Ch=0x0 Portnum=3 rxtime=1703244432)
DEBUG | 11:27:12 60 [Router] handleReceived(REMOTE) (id=0x4ee8f68d fr=0x00 to=0x6c, WantAck=0, HopLim=0 Ch=0x0 Portnum=3 rxtime=1703244432)
DEBUG | 11:27:12 60 [Router] Module 'position' wantsPacket=1
INFO  | 11:27:12 60 [Router] Received position from=0x0, id=0x4ee8f68d, portnum=3, payloadlen=13
DEBUG | 11:27:12 60 [Router] Incoming update from MYSELF
DEBUG | 11:27:12 60 [Router] Setting local position: latitude=0, longitude=0, time=1703244380
INFO  | 11:27:12 60 [Router] POSITION node=1f37e46c l=13 latI=0 lonI=0 msl=0 hae=0 geo=0 pdop=0 hdop=0 vdop=0 siv=1 fxq=0 fxt=0 pts=1703244380 time=1703244380
INFO  | 11:27:12 60 [Router] updatePosition REMOTE node=0x1f37e46c time=1703244380, latI=0, lonI=0
DEBUG | 11:27:12 60 [Router] Node status update: 7 online, 7 total

and Andoid 
INFO  | 11:27:30 77 To Radio onwrite
DEBUG | 11:27:30 77 **PACKET FROM PHONE** (id=**0xf3aa9962** fr=0x00 to=0x6c, WantAck=0, HopLim=5 Ch=0x0 Portnum=3 priority=10)
DEBUG | 11:27:30 77 Enqueued local (id=0xf3aa9962 fr=0x00 to=0x6c, WantAck=0, HopLim=5 Ch=0x0 Portnum=3 rxtime=1703244450 priority=10)
DEBUG | 11:27:30 77 [Router] Add packet record (id=0xf3aa9962 fr=0x00 to=0x6c, WantAck=0, HopLim=5 Ch=0x0 Portnum=3 rxtime=1703244450 priority=10)
DEBUG | 11:27:30 77 [Router] handleReceived(REMOTE) (id=0xf3aa9962 fr=0x00 to=0x6c, WantAck=0, HopLim=5 Ch=0x0 Portnum=3 rxtime=1703244450 priority=10)
DEBUG | 11:27:30 77 [Router] Module 'position' wantsPacket=1
INFO  | 11:27:30 77 [Router] Received position from=0x0, id=0xf3aa9962, portnum=3, payloadlen=17
DEBUG | 11:27:30 77 [Router] Incoming update from MYSELF
DEBUG | 11:27:30 77 [Router] Setting local position: latitude=478892456, longitude=333090801, time=1703261574
INFO  | 11:27:30 77 [Router] POSITION node=1f37e46c l=17 latI=478892456 lonI=333090801 msl=87 hae=0 geo=0 pdop=0 hdop=0 vdop=0 siv=0 fxq=0 fxt=0 pts=0 time=1703261574
INFO  | 11:27:30 77 [Router] updatePosition REMOTE node=0x1f37e46c time=1703261574, latI=478892456, lonI=333090801
DEBUG | 11:27:30 78 [Router] Node status update: 7 online, 7 total

Shmarteee, 2023-12-24:
Same problem with iOS 17.2.1 and Heltec 32 V3 = No GPS or Lat/Lon being pushed to device or read by app. 

All GPS/Location permissions in iOS and Meshtastic app have been allowed ('Always' access). 

[image]
[image]

geftactics, 2023-12-24:
@Shmarteee yours does actually look like a different issue - the coordinates in the app are showing as 0,0 - whereas I can see the correct coordinates in my iOS app!

postcatz, 2023-12-24:
@geftactics my app said the correct location yesterday before i updated(still wasnt published to the mesh tho), now it shows 0,0 too - i deleted the app and reinstalled it but it now still says location 0,0

postcatz, 2023-12-24:
update: issue is already fixed and will most likely be released soon

closed 2023-12-25 by garthvh