meshtastic/Meshtastic-Apple #438: 🐞 [Bug]: Distances in node list are to lat,lon 0,0 instead of actual location

gitToad, 2023-12-22:
### Firmware Version

2.2.15

### What did you do?

I have a bunch (15) Meshtastic devices on our mesh, some T-Echos, some T-Beams, some Garth devices. Some have GPS, a few do not.
using various (relatively recent 2.2.12 and up) firmware versions and app version 2.2.15
Connecting to any node, looking at node list in the app and specifically the distance in miles to the node shown.

### Expected Behavior

Expect to see the distance from myself (either BT connected device or phone/laptop, not sure which one is intended to be used of the two) to each node in the node list that has transmitted at least one position packet.

### Current Behavior

[image]

In the device list (on Mac & iOS), devices for me appear 7956 miles away (I am in SF Bay Area and this is the distance to lat, lon = 0,0) This is regardless of whether I am connected to a node that HAS or HAS NOT got GPS onboard. It also does not matter whether the app is sharing location or not. 

Both the laptop and phone themselves know their location correctly (ie I can go into Google Maps and the blue dot is in the correct place) so I suspect this is a bug in the app, where the "self" location is somehow 0,0

### Participation

- [ ] I am willing to submit a pull request for this issue.

ERF1994, 2023-12-23:
[image]

same problem, my node T-beam v1.1, ios17.2.1 updated Meshtastic app. When I switch to legacy map the distance on map displayed right, but node list still wrong.

gitToad, 2023-12-23:
Possibly related (ie if app erroneously thinks it is at 0,0 then those distances make sense)
Seeing lat lon 0,0 reported on meshtastic app 
• on ios 17.2.1, app version 2.2.16 as well as 
• Mac OS 14.2.1 app version 2.2.15
[image]

guillochon, 2023-12-23:
GPS locations appear to be broken in several versions of iOS for me (Tried iOS 17.1.2, 17.2.1, and iPad OS 17.1.1) on the latest version of the Meshtastic app. I see my GPS being 0, 0 regardless of whether I am paired to a device or not.[image]

ayypablopablo, 2023-12-23:
iPad mini 6th gen on OS 16.5.1  Location settings show the purple location checkmark (active) for Meshtastic.  Meshtastic sees 7 satellites and coordinate positions

iPhone 15 Pro on OS 17.2.1 
No active location (purple location checkmark) at any point with Meshtastic open. Meshtastic displays 000000,00000 coordinates and specifies estimated 1 satellite.

Findings for both devices above occurred at the same time. Meshtastic version 2.2.16

mesh100A, 2023-12-24:
I'm having the same issues. With IOS 16.7.4 and the latest Meshtastic APP. I see in app settings 7 sats and the correct co-ordiantes for my location but my Heltec V3 does not show KM and just a ? in the circle GPS. So in this setup the App is taking the Coorodiantes from the mobile to app but no displaying them on the unit. When I move to IOS 17.2.1 and latest app nothing works. No co-ordiantes in mash appo and still no GPS data on V3. Meshtastic displays 000000,00000 coordinates

mesh100A, 2023-12-24:
[image]

mesh100A, 2023-12-24:
[image]

mesh100A, 2023-12-24:
[image]
This is the display on either ios 16.7.4 or 17.2 and 17.2.1 If you roll back the app to 4 revisons it works I have been told

I3rixon, 2023-12-24:
Last version of App 2.2.16 
[image]

postcatz, 2023-12-24:
just built the all myself and location and location reporting work fine!

postcatz, 2023-12-24:
the isssue is fixed in 2.2.17 and will most likely be released very soon

Kiloa1, 2023-12-24:
I have the same issue as above. The app was working perfectly and sharing data to the node.
Yesterday I removed and reinstalled the app from fresh, the coordinates were no longer being sent to the Heltec board. I could see the coordinates in the app. 
To try and resolve the issue i decided to update IOS to the latest revision and suddenly the issue was worse.
No the coordinates!  the same issue as others are reporting. Basically they are not being picked up or sent to the board.

postcatz, 2023-12-24:
@Kiloa1 it's already fixed and will be released soon

mesh100A, 2023-12-24:
> @Kiloa1 it's already fixed and will be released soon

thank you so much, cant wait

closed 2023-12-25 by garthvh