meshtastic/Meshtastic-Apple #439: 🚀 [Feature Request]: add "last seen" age to node and map display

gitToad, 2023-12-23:
### OS

iOS, iPadOS, macOS

### Description

Saw this on the Android app and found it quite useful: on the Map display, it shows along with each node's name and icon, the time ago it was last seen ("3d", "20m", "now" etc.) This helps with figuring out how much to trust that location - I'm on ios/mac and found myself multiple times scratching my head or cursing a T-Echo for a really wrong location only to find out it was quite old and stale.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

It would be nice to have that last message "age" display to the right of the date/time in the node listing display (i think there is real estate?) 
as well as in the map display(s) to the right or under each node name

garthvh, 2023-12-23:
Positions that are considered online (seen in the last 15 minutes) are animated.

garthvh, 2023-12-23:
The time is also available in the position popup

closed 2023-12-23 by garthvh

gitToad, 2023-12-23:
excellent, did not realize the throbbing nodes meant recent - thank you !