meshtastic/Meshtastic-Apple #446: 🐞 [Bug]: Fetching data from node incomplete even though it says complete ?

gitToad, 2023-12-26:
### Firmware Version

Mac app Version 2.2.17 (754)

### What did you do?

Not sure how to title this. Or whether it is a bug or feature enhancement request. 

Repeatably, after disconnecting from a node and then connecting to another node via Bluetooth on my Mac (M1 Laptop, 14.2.1), the node connects, "communicating with device" animation, and then 2 things happen:

1. firmware version always shows as "Unknown"
2. switching to "Settings" screen, I get some orange warnings and cannot yet edit device parameters

BUT (this is the workaround also): when switching back to Bluetooth screen, the firmware version now shows correctly and after switching back to "Settings" once more, the warning message has disappeared and the device paramters are editable.

This behavior appears independent of what kind of device connects (ie I have seen it on ESP32 Heltec v3's as well as RAK4630's)

### Expected Behavior

It feels like the first time after connecting, something is left uninitialized or incomplete and after round tripping back to the bluetooth screen and settings, things work as expected.

### Current Behavior

Here is a picture of what happened if that makes it any clearer:
[image]

Feel free to change from a bug to a feature request (or close it) if this is just a minor UI nuisance. Filing it just in case it points to some potential unintended behavior.

Thank you !

### Participation

- [ ] I am willing to submit a pull request for this issue.

gitToad, 2023-12-26:
Here is a picture of the sequence of events if that makes it any clearer:
[image]

garthvh, 2024-01-15:
There is a fix for this in TestFlight, will be in the next release

closed 2024-01-15 by garthvh