meshtastic/Meshtastic-Apple #447: 🐞 [Bug]: Location from phone being pushed to node too frequently

guillochon, 2023-12-27:
### Firmware Version

2.2.16.1c6acfd

### What did you do?

I took these three photos in quick succession (over 5 seconds) after a fresh restart of a node. As you can see from the app settings the position shouldn’t be updating this frequently. I had turned off my other node entirely during this test, it’s just one node and the phone. The node became unresponsive to button presses within about 20 seconds of power on.

[image]
[image]
[image]
[image]

### Expected Behavior

Location should only be pushed once every 30 seconds (or whatever is set in the App Settings)

### Current Behavior

Location appears to be pushed every few seconds, as indicated by the wildly fluctuating arrow.

### Participation

- [ ] I am willing to submit a pull request for this issue.

guillochon, 2023-12-29:
This is still affecting me. Can I provide some debug info? Any sort of acknowledgement from the team that the issue exists?

guillochon, 2023-12-31:
I was able to resolve this by clearing the app settings and then reinstalling the app entirely. Hasn't recurred since then, but I'll keep this open for now to see if others encounter the same thing.

guillochon, 2024-01-08:
Was working fine for about a week, but have run into this issue again. I have since upgraded the firmware on the node to 2.2.17. Turning off "location sharing" in the App Settings seems to make the node responsive, but as soon as I turn it on, it's unresponsive. Dunno if that's because the location is indeed being shared too frequently or what.

garthvh, 2024-01-14:
There is a fix in test flight preventing 0 as the timer interval.

closed 2024-01-14 by garthvh

guillochon, 2024-01-14:
Thanks @garthvh!