meshtastic/Meshtastic-Apple #449: 🐞 [Bug]: Traceroute no longer works

GaryR-12, 2023-12-30:
### Firmware Version

2.2.15.31c4693

### What did you do?

Used to be able to request a traceroute in the ios app and get the result in the Mesh Log. For the last 3 weeks (after updating the app) Traceroute is now under NODES instead of MESSAGES and does not work anymore. The log will only show a traceroute that is received directly, no other traceroute responses show in the log. A buddy reports the very same behavior on his android device.

### Expected Behavior

Expect to see the radio numbers which hopped the message to it's destination radio.

### Current Behavior

The log will only show a traceroute that is received directly

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2023-12-30:
There have been firmware changes that have adjusted over the last year, I also get direct connections from spots I used to have hops.  The trace route info is still in the mesh log as well.

closed 2023-12-30 by garthvh

GaryR-12, 2023-12-30:
I have not gotten any direct connections from nodes that used to have hops. All direct connections are as expected from nearby radios. Distant Nodes that I can message with through a router no longer return any traceroutes in the log.

GaryR-12, 2024-01-14:
@garthvh  Traceroute is back after updating to 2.2.16