meshtastic/Meshtastic-Apple #450: 🐞 [Bug]: Nodes swich to UNKNOWN AGE seamingly at random

GaryR-12, 2023-12-30:
### Firmware Version

2.2.15.31c4693

### What did you do?

In the Nodes display, Several nodes in my mesh keep switching from a date and time to the UNKNOWN AGE label.

### Expected Behavior

Used to seeing each node display the date and time of it's last contact.

### Current Behavior

several nodes show UNKNOWN AGE

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2023-12-30:
It is not random, a packet came in with a time that is 5 years old or greater. I keep finding pieces of this and I think some of the recent firmware updates may help.  I have a fix for any times that are 0 in the next release.  It does not really work to fix up times in the app as they come right back in over the mesh or when you connect again.

GaryR-12, 2023-12-30:
The three nodes that I currently have that are showing UNKNOWN AGE were active in the last 1-7 days.

garthvh, 2024-01-14:
Yeah the time come from the node, you can open time issues in the firmware repo, I just display the data.

closed 2024-01-14 by garthvh