meshtastic/Meshtastic-Apple #453: 🐞 [Bug]: Position Configuration issue with iOS app

oripaamoni, 2024-01-04:
### Firmware Version

2.2.16

### What did you do?

Using the iOS app the GPIO assignment is a drop down list that doesn’t include all the pins. After successfully configuring the gps with the android app if you then access the position configuration with the iOS app the gps ceases to function. Even when assigning and connecting to pins listed in the iOS app the GPS would work until the esp32 was rebooted. Strange because it would  update the time from GPS but also report “no gps module” and never get its position despite the GPS module indicating that it had a GPS lock. Once the GPS is configured with the android app everything works great as long as you don’t access the position menu with the iOS app. For reference I am using a pair of HELTEC Lora v3 with NEO-M6 based module. 

### Expected Behavior

Should be able to see text any pin assignment available with the iOS app.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-01-04:
What pins do you think are missing? What board are you using?

oripaamoni, 2024-01-04:
Using a Heltec V3, using pines 45 and 46. The iOS app only shows up to pin 45 in the list.

LanWolf, 2024-01-10:
For LiliGo T3S3 i miss pin 46 in The iOS app.
Also configuring seems to have an issue with 2.2.17, keeps saying no sats after configure. 
When going back to 2.2.15 and configure gps with pin 45 via iOS app it works, even after updating to 2.2.17 without erasing.
Changing anything in the gps menu when on 2.2.17 breaks it again.

garthvh, 2024-01-14:
Extra pins have been added in TestFlight, will go out in the next version

closed 2024-01-14 by garthvh