meshtastic/Meshtastic-Apple #454: 🐞 [Bug]: Y2K: Nodes show the date  10/1/2054

GaryR-12, 2024-01-06:
### Firmware Version

2.2.16

### What did you do?

Nothing. Noticed it after new years

### Expected Behavior

correct date

### Current Behavior

After new years, about 50% of my nodes (4) showed a last heard date of 10/1/2054

### Participation

- [ ] I am willing to submit a pull request for this issue.

closed 2024-01-14 by garthvh

GaryR-12, 2024-01-17:
@garthvh  What does "Apple has fixed these issues" mean?
The date on one of my busy nodes just went back to last November.
Several nodes just went to 6/3/2083

garthvh, 2024-01-17:
Likely your techo then, if you have a bad time source on your mesh it is going to spread. Make sure your os and nodes are updated.