meshtastic/Meshtastic-Apple #455: 🚀 [Feature Request]: add "Request Position" to node popup / long click menu

gitToad, 2024-01-07:
### OS

iOS, iPadOS, macOS

### Description

Requesting position would be useful if for memory/mesh traffic reasons you want to keep periodic position packet pushes infrequent.
But available "on demand" 
The Android app currently has this in the pop up menu. And while I realize that different people design, write and maintain the Android and Apple apps, it would be nice to have that feature on the Apple side too. 
Not a must have, but nice.

### Participation

- [ ] I am willing to submit a pull request for this issue.

gitToad, 2024-01-07:
Actually isn't that a "push location" just like the periodic push already done by devices ?
The Android functionality (and the one I was referring to) is a "pull" or "poll" "Request location" from another device. Ie if command post wants to know where someone is right now (without having to ask them, them having to be available to pick up the phone, replying with the push)

garthvh, 2024-01-08:
Nope, you have to send your position to receive a position back from another node, I think the way the iOS app is set up is much clearer that you are sending your position.