meshtastic/Meshtastic-Apple #456: Resend failed message button

dmthepm, 2024-01-07:
### OS

iOS

### Description

When a message fails to send due to max transmission reached, I would love to be able to long press and hit resend. Maybe you can leave the record of the failed send there but just populate and send a new message. 

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-01-19:
Both phone apps currently allow you to copy the message and resend it yourself.

This has been on my list for a while, but it is hard to know how to handle it in a mesh friendly way, acks can take quite a while.
[image]

drewzh, 2024-01-31:
I would find this incredibly handy. I'm sometimes on the move and need to send out 'ping' messages to test propagation. Having to retype, even 'ping' each time I need to send a test, is quite tedious. One other potential fix for this (in my use case at least), would be to provide an ability to send a canned message, which requires only 2 clicks from the user.

garthvh, 2024-02-07:
With messages taking up to 10 minutes the ability to copy and paste a message will remain the solution.

closed 2024-02-07 by garthvh

drewzh, 2024-02-20:
> With messages taking up to 10 minutes the ability to copy and paste a message will remain the solution.

@garthvh I understand the desire to not allow re-transmission of a message that may already be in play (and why would you, the system does that for you), but when the max-transmission limit has been reached, would it not be nice to give the user a quick retry option, perhaps with a small link that's easy to click - or still really useful, an option on the context menu to re-send?

For me, copying and pasting whilst I'm on the go is sometimes very cumbersome.

thebentern, 2024-02-20:
@drewzh this has been already with implemented with #486

garthvh, 2024-02-20:
> @garthvh I understand the desire to not allow re-transmission of a message that may already be in play (and why would you, the system does that for you), but when the max-transmission limit has been [...]

https://support.apple.com/en-ca/guide/iphone/iph6d01d862/ios