meshtastic/Meshtastic-Apple #460: 🚀 [Feature Request]: Allow adding an existing channel

dustin, 2024-01-13:
### OS

iOS, iPadOS

### Description

For an existing meshtastic user, there is currently no way to add an existing encrypted channel.

A channel requires a PSK (pre-shared key) to be able to send and receive messages, but the iOS UI does not have a means of supplying a key that was created for an existing channel.

In practice, this means that if you have a typical setup (a private primary, public LongFast, an admin channel, etc...) and wish to join a group, the only option available to join this group is to scan a QR code which will overwrite all of your channel settings.  Ideally, one would be able to communicate with one group without losing communication with another.

## Concerns

There is a concern that allowing a key to be pasted in might result in someone generating a less secure shared key.  This is certainly possible, but it's already possible via QR code.  While we should encourage our best security practices for end users, which means using a secure mechanism for generating a PSK, we shouldn't make it impossible for them to communicate with others out of concern that the key wasn't controlled well.

## Precedent

To add an existing channel today, one may use the Android client, web client, or python cli, so there is precedent (and workarounds for some).

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

This has come up in discussion a few times with users who have been limited by it.

ianmcorvidae, 2024-01-13:
As this has been discussed in other channels, the best idea for algorithm/UI that I've had is the following, though I'm confident it could use more refinement:

* When a QR code is scanned, if the non-channel-related settings (e.g. modem preset) match the existing settings, prompt the user to ask if they would like to replace their current settings or only add channels. 
* If they say "replace", continue as currently implemented.
* If they say they want to only add channels, present an interface that allows them to select a set of channels from the existing settings & the QR code, up to the maximum allowed channels, to become their new channel list (and if possible, allow reorganizing the list or otherwise selecting the order, particularly in order to set the primary channel -- presumably, if such an interface exists, matching an existing "rearrange channels" sort of interface).
* Once selected/confirmed, determine from those selections a new set of settings & channels and apply it as usual.

This would allow QR codes that only 'export' a subset of channels to be scanned by other users in order to add channels without requiring a copy-pastable or typable field that's prone to transcription errors nor overwriting existing channel settings.

garthvh, 2024-01-14:
This would require key derivation function in order to be secure, which would then not work for pasting simple passwords.   I don't believe there is a secure way to allow users to paste strings in as AES keys.

dustin, 2024-01-14:
> I don't believe there is a secure way to allow users to paste strings in as AES keys.

It's not clear what "secure" means in this context.

This is a Pre-Shared Key, so in order for two people to communicate on a channel, they have to be able to share a key.  Without a way to do that, it's certainly more secure as messages sent to a channel would only be readable by the person who created the channel.  That's not useful, though.

Receiving a channel configuration via QR code is equivalently secure to receiving one as text and passing it in.  Both allow a key to be generated insecurely, passed through an uncontrolled channel that might be compromised, and then applied without additional validation as to the provenance of the key.  Using a QR code might provide a better user experience, but it doesn't provide additional security.  In this case, there's no way to add a channel via QR code anyway, so that's a separate issue.

Without a means of sharing a pre-shared key, there's no way for an iOS user to join an existing channel that might have a community that the user would like to be part of (e.g., a first response team).

Similarly, being able to do in-band revocation of PSKs would greatly enhance security.  For example, if I have a small channel of people I trust and guard that key carefully and a larger group channel that has been compromised and needs to be rekeyed, it'd be possible to propagate that new key over meshtastic directly via circles of private channels.  In this context, not being able to paste in a PSK makes the system strictly less secure.

Related:  When creating a channel via the web app (which didn't offer to generate a key for me), I used this reference for securely generating PSKs:  https://cloud.google.com/network-connectivity/docs/vpn/how-to/generating-pre-shared-key – they do agree (and nobody is arguing) that best practices require generating a key using a secure mechanism.  I didn't really trust the browser one when I set mine up, so I used openssl directly.

dustin, 2024-01-14:
> This would allow QR codes that only 'export' a subset of channels to be scanned by other users in order to add channels without requiring a copy-pastable or typable field that's prone to transcription errors nor overwriting existing channel settings.

Yeah, or could even just have a QR code for adding a channel.  That might be easier, I don't know.  I just want to add a channel.  :)

garthvh, 2024-01-14:
Not planning on allowing users to paste in keys, these keys need to be generated randomly, not created from arbitrary strings.

garthvh, 2024-01-14:
This is an easy win for security, and prevents users from using compromised passwords. Not willing to compromise security for a power user edge case, sorry.

closed 2024-01-14 by garthvh