meshtastic/Meshtastic-Apple #462: 🚀 [Feature Request]: DM channels should be visible and selectable

dustin, 2024-01-17:
### OS

iOS, iPadOS

### Description

When sending a DM, the channel is automatically selected in such a way that the user can neither influence nor even determine over what channel the message was sent.  This leads to messages being sent insecurely or even part of a conversation alternating between secure and insecure delivery.

The mechanism appears to be related to a channel negotiation for interacting with a node.  The channel negotiation doesn't necessarily yield desirable results for the user (e.g., perhaps there are multiple valid channels, but a user wants to make sure it goes over the one where the key has the smallest circulation).

The mechanism also fails sometimes defaulting to the 0 channel which may either not work, or expose the users's messages to unintended recipients.

Similarly, when receiving a message, it doesn't seem possible to determine which channel the messages was received on, so one can't determine whether a messages may have been more broadly exposed.

Ideally, the channel could be locked in from the sender and visible on the client where the DM view shows that these things are happening with the intended securely.

A less-desirable alternative (but perhaps still important thing to do for other reasons) might be to lock a node to a particular channel for default communication since failing to zero is rarely desirable.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-01-17:
DM's are sent over the primary channel a fix for the channel being off sometimes is going to be in firmware 2.2.19.

GUVWAF, 2024-01-17:
In my opinion, channel *selection* is not something that should be exposed to the user, as they may not know which channel can be used to reach the node. The firmware at least ensures that a valid channel is chosen (after https://github.com/meshtastic/firmware/pull/3094 is released). It also can’t be locked in since you don’t know when the other nodes change their channels.

Multi-channel DM handling was only merged because it was agreed upon that it’s not 100% secure and that was also not the goal: https://github.com/meshtastic/firmware/pull/2384#issuecomment-1483907386

closed 2024-01-17 by garthvh

dustin, 2024-01-18:
If a user configured a channel that didn't work, they wouldn't get an ACK that the message had been received.

Right now, a message can (and does) go through in the clear and there's no way anyone could know about it.

This seems like a rather bad security issue and doesn't seem particularly hard to not make a bad security issue.

It's also not clear why exposing the channel to the user is a inherently bad.  If my partner and I have a secure channel we wish to communicate over, why would it be wrong for us to specify that in our communication?  If that channel is unavailable, it would be better to know that than to just communicate in the clear.

riddlemd, 2024-01-18:
I have to agree that this is a poor implementation that leads to security risk. There should never be a situation where something is flipping between encrypted and not encrypted without presentation to the user.

GUVWAF, 2024-01-18:
> If a user configured a channel that didn't work, they wouldn't get an ACK that the message had been received.

That sounds like pretty bad UX, but since the Apple apps then show the appropriate “No channel” error message, it at least gives a clue what the user should try next. I believe at least the default should be “let the firmware handle it”. Maybe a user should be able to overwrite it, but I don’t know how that would work in the UI.

> I have to agree that this is a poor implementation that leads to security risk. There should never be a situation where something is flipping between encrypted and not encrypted without presentation to the user.

You’re expecting something from a feature that wasn’t even designed to be secure and then call it a “poor implementation”. Nothing will be flipped between encrypted and not encrypted if you don’t use the public channel (which is actually encrypted but with a known key). Compromises have to be made in a low-bandwidth mesh running on microcontrollers.

dustin, 2024-01-18:
I don't think it's a poor implementation, but I do think the UX should be improved to make it easier for someone to know when a channel is being chosen that shouldn't be.

I do understand that it's possible to go to the node list, find the node in question, see whether it's negotiated a channel, and then go back to the DM list and send a message assuming it's not been updated.  I do think we can provide a better UX.

Are people actually strongly opposed to the user being able to know which channel would be used for sending a DM?  Or to know what channel a DM was sent (or received) on?

Regarding the user being able to specify a channel, in chat, GUVWAV said:

> ’m not into UI but if someone implements this in a way that it “just works” in case you don’t want to override it, I would be fine with it.

This is consistent with my expectations.  I would expect DM to "just work" in most cases, but if wanted to make sure DMs to a certain recipient would only go over a specific channel that I'd created, I'd like to be able to request that, understanding the caveats.

Alternatively, someone else proposed marking channels as acceptable or not acceptable for DMs.  Allowing users to influence security decisions made on their behalf sounds like a good idea.  Flipping between a "fail open" model and a "fail closed" model is quite within reach.

Other members suggested you should either not expect security, or not have a channel configured you didn't want DMs going over.  If this is truly the project's opinion, it'd be good to be clearer about this in the documentation.  But I think doing better is easily within reach.

Of course, this bug was closed before any of the discussion…