meshtastic/Meshtastic-Apple #466: 🐞 [Bug]: App crash when tapping on Mesh Map

r-vdz, 2024-01-20:
### Firmware Version

2.2.17

### What did you do?

Launch iOS app. Click on 'Mesh Map'.

### Expected Behavior

Map loads.

### Current Behavior

App crashes.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

iPhone 14 Pro
iOS 17.2.1
Meshtastic version 2.2.19

hutchisr, 2024-01-20:
I also have this problem on the iOS version (iPhone 15 Pro, same iOS version), the Mac version is fine. Anecdotally seems to happen when the node list gets above around ~35.

garthvh, 2024-01-20:
[image]

If you can enable crash reporting I can take a look, I have 100+ nodes and some very large position logs and routes on my map so not a number thing for a crash (may slow down or struggle to draw history pins)

r-vdz, 2024-01-20:
Done. 

Weirdly, after this I tapped on maps again, it loaded but as I zoomed in from the globe it crashed. Launched the app again, went to Maps, it looked like it was starting to load the map and then crashed. 

I then turned on 'Use Legacy Mesh Map' and this seems to work fine.

hutchisr, 2024-01-20:
There aren't any Meshtastic files from today (most recent one is the 17th) despite multiple crashes (including after clearing app data). I have a ton of Jetsam Events that list Meshtastic as the largest process though. Seems like something is causing a lot of memory usage.

[JetsamEvent-2024-01-20-001250.ips.gz](https://github.com/meshtastic/Meshtastic-Apple/files/13999313/JetsamEvent-2024-01-20-001250.ips.gz)

garthvh, 2024-01-24:
Closed by https://github.com/meshtastic/Meshtastic-Apple/pull/465 Updated in test flight, submitted to the App Store

closed 2024-01-24 by garthvh