meshtastic/Meshtastic-Apple #468: 🚀 [Feature Request]: Effective Channel in LoRa Config Settings

Xaositek, 2024-01-23:
### OS

iOS, iPadOS, macOS

### Description

There is likely a small number of new users which may find the logic of "LoRa Channel Number" confusing. While the Tips page (https://meshtastic.org/docs/configuration/tips) does a good job of explaining when you need to define a channel versus not; there is a subset of users who generally enjoy confirmation of effective configuration.

**FR:** Add a line called "Effective LoRa Channel Number" or update the original "LoRa Channel Number" to truly reflect which channel is currently being used.

**Reasoning:** If someone wasn't running the default, so they manually defined channel 20, but my app showed 0 - the natural inclination is "I need to set this to 20 so they match" which is unnecessary.

### Participation

- [ ] I am willing to submit a pull request for this issue.

riddlemd, 2024-01-23:
This is an issue that spans more than just iOS. The auto selecting of LoRa channel (lch) via primary meshtastic channel (mch) confuses a bunch of people.

I've talked to a lot of folks that set LongFast as their secondary and switch their primary and can't figure out why LongFast no longer works (I know it's explained in the docs to set lch manually in that case, but people are just not conceptualizing it.)

garthvh, 2024-01-26:
This is the output of a default US region radio

INFO  | ??:??:?? 1 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 1 Radio channel_num: 20
INFO  | ??:??:?? 1 Radio frequency: 906.875000
INFO  | ??:??:?? 1 Slot time: 42 msec
INFO  | ??:??:?? 1 Set radio: final power level=20
DEBUG | ??:??:?? 1 Current limit set to 100.000000
DEBUG | ??:??:?? 1 Current limit set result 0
INFO  | ??:??:?? 1 RF95 init result 0
INFO  | ??:??:?? 1 Frequency set to 906.875000
INFO  | ??:??:?? 1 Bandwidth set to 250.000000
INFO  | ??:??:?? 1 Power output set to 20

garthvh, 2024-01-26:
The lora channel number does not show in --info because it is 0

garthvh, 2024-01-26:
Now I changed the primary channel name to NewChan

INFO  | ??:??:?? 1 Radio channel_num: 98
INFO  | ??:??:?? 1 Radio frequency: 926.375000
INFO  | ??:??:?? 1 Slot time: 42 msec
INFO  | ??:??:?? 1 Set radio: final power level=20

The lora channel number still does not show in --info because it is 0

closed 2024-01-26 by garthvh