meshtastic/Meshtastic-Apple #476: 🐞 [Bug]: Incorrect visualization for the configuration of the detection sensor.

OE4DEA, 2024-02-02:
### Firmware Version

2.2.17

### What did you do?

Incorrect visualization for the configuration of the detection sensor.

[image]

### Expected Behavior

Rename to "Detection Sensor".

### Participation

- [ ] I am willing to submit a pull request for this issue.

OE4DEA, 2024-02-02:
> What language are you using?

I use German.

garthvh, 2024-02-06:
closed by https://github.com/meshtastic/Meshtastic-Apple/pull/472 

They are just the English versions so they will need translation.

closed 2024-02-06 by garthvh