meshtastic/Meshtastic-Apple #477: 🐞 [Bug]: More than 35-ish nodes causes iOS to terminate the app for using too much memory when using the map view

hutchisr, 2024-02-03:
### Firmware Version

2.2.19.8f6a283

### What did you do?

1. Connect to a meshtastic device using BLE that knows of over 30 nodes
2. Open map screen

### Expected Behavior

The app should stay open.

### Current Behavior

iOS closes the Meshtastic app immediately when viewing the node map

### Participation

- [X] I am willing to submit a pull request for this issue.

### Additional comments

Current workarounds:
* Use legacy map mode.
* Rebuild the app with the "Increased Memory Limit" entitlement.

garthvh, 2024-02-03:
35 is pretty low, I had slowdowns at 155, currently have 78 and am not seeing any memory pressure.

garthvh, 2024-02-03:
Do you have a lot of favorites and node history on?  How big are your position logs?

hutchisr, 2024-02-03:
Only 3 favorite. Not quite sure what you mean by node history? I'm not using store and forward. Position logs don't seem to be a factor since even after cleaning app data the issue persists. Is there any way I can export my node list that might provide some more insight?

garthvh, 2024-02-07:
Not seeing over a gig of memory used any more, the maps do use some memory to display hundreds of pins and history

closed 2024-02-07 by garthvh

hutchisr, 2024-03-02:
@garthvh why did you close this? It's still an issue.