meshtastic/Meshtastic-Apple #479: 🐞 [Bug]: Keyboard gets very laggy after a day or so

BjornB2, 2024-02-06:
### Firmware Version

2.2.21

### What did you do?

Just use the meshtastic app with a bit of chatting

### Expected Behavior

Keyboard perfomrance being normal

### Current Behavior

After a day or so the keyboard becomes very laggy, but only in the Meshtastic iOS app. In other apps all is fine. If I reinstall the app or use the "Clear app data" button (I found that out later) the keyboard functions normally for a day or so.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

iPhone 15 Pro
iOS 17.2.1
Meshtastic App 2.2.20 (777)

Firmware version on top is the connected Meshtastic node (2.2.21)

closed 2024-02-07 by garthvh

Austinpayne, 2024-02-13:
Should be fixed by https://github.com/meshtastic/Meshtastic-Apple/pull/484

BjornB2, 2024-02-13:
Awesome. And it was not just old phones. I have a iPhone 15 pro and after a while it became unworkable.