meshtastic/Meshtastic-Apple #483: 🐞 [Bug]: V2.2.21(787) Unread Badge with messages to 'admin' channel

RejectH0, 2024-02-12:
### Firmware Version

2.2.21

### What did you do?

From my Android device, I successfully sent a message in the "admin" channel that I created.   The iOS Meshtastic app refuses to display the "admin" channel in the "Messages" bottom-screen tab, so there is an unread message in an inaccessible channel, thereby causing the badge notification to persist without the ability to clear it.

### Expected Behavior

I should somehow be able to clear the notification badge indicating an unread message in an inaccessible channel.

### Current Behavior

I cannot get rid of the red badge notification "1" indicating an unread message inside of an inaccessible channel.  

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Yeah, it was my dumb mistake for sending a message in the "admin" channel, but the Android Meshtastic client allows this, where the iOS application does not allow any access to the "admin" channel.  I'm brand-spankin'-new to Meshtastic, so of course I broke something the very first day I started playing with it.  This has been happening to me since I was a child; I've been finding bugs in all kinds of programs and devices since I was like 6 years old.

I would submit screenshots, but they will not be helpful.  Everything I've described is exactly as it's happening.  I have a red "1" badge on the Meshtastic App in the Springboard, as well as a red "1" badge on the "Messages" tab at the bottom of the iOS Meshtastic app.  Tapping on it brings me to "Channels" and "Direct Messages," where "Channels" has a "1" at the right side.  I tap on that, and then I see three of the four channels I have configured.  The "admin" channel is not listed here, and I therefore cannot clear the unread badge notification.

hatesyourface, 2024-03-08:
Possible suggestions: 
1. Show the admin channel, but make it read-only
2. Above, plus an app setting to show or hide the admin channel (hidden by default)
3. A "mark all as read" button above the channel list, which would also clear the notification.

AssetBurned, 2024-03-14:
The problem here is that the App enforces a functionality that is clearly available outside of the Apple App. The Python tool, the Android App and the Web App all allow normal interaction with the admin channel as it would be a normal channel. So either the App needs to loosen the grip on that channel, or it needs to be enforced cross platforms that this channel is not accessible for messages. And enforcing that would then be and firmware thing, right?

closed 2024-04-09 by garthvh