meshtastic/Meshtastic-Apple #489: 🐞 [Bug]: Display error on Device Metrics Log and Environment Metrics Log

moresecode, 2024-02-16:
### Firmware Version

2.2.21

### What did you do?

Hardware use: RAK4631 + RAK1906 (BOSCH BME680)

Nodes > tap on my node > Device Metrics Log
Nodes > Environment Metrics Log

### Expected Behavior

Display readable information.

### Current Behavior

Those green and yellow dots are all over the places.
[image]

Temp reading has too much decimal places.
[image]

### Participation

- [X] I am willing to submit a pull request for this issue.

garthvh, 2024-02-16:
Channel utilization and airtime bounce around a lot more.

garthvh, 2024-02-17:
There is another option for "width" for the measurement values, the next App Store release will use narrow for temperature.  The scatter plot for channel utilization is the the right visualization for something than can bounce around so much, you can see in that data there are potentially 10% or more jumps every minute.

closed 2024-02-17 by garthvh