meshtastic/Meshtastic-Apple #507: 🚀 [Feature Request]: Manually set GPS coordinate and not use the phones location

AssetBurned, 2024-02-20:
### OS

iOS, iPadOS, macOS

### Description

Option to set a fix location via Map and/or via numeric values.

For privacy reasons you may want to be able to set a location that is not based on your phones location, but still want to provide a rough location to make others aware of a node.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-02-20:
I will support the new location ambiguity when it is released, but not spoofed locations.

closed 2024-02-20 by garthvh

AssetBurned, 2024-02-20:
so if the setting is not touched in the app. will it respect the config set via the command line tool?