meshtastic/Meshtastic-Apple #510: 🐞 [Bug]: Map broken when viewing trace route between 2+ nodes

drewzh, 2024-02-20:
### Firmware Version

2.2.22

### What did you do?

Perform traceroute to client node that goes via an intermediate node. Al nodes in question have location data seta 

### Expected Behavior

Should show line between client, router and client on the map

### Current Behavior

Map fails to load

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

[image]

garthvh, 2024-02-20:
The map only shows if we have a recent position from each node

drewzh, 2024-02-20:
> The map only shows if we have a recent position from each node

As mentioned, all nodes have location data set. The client has GPS fix as of only minutes earlier, the router location is static and recently reported/on the map and the endpoint client node has a static location set/reported earier and viewable on the map.

drewzh, 2024-02-20:
Just a small note also, though more anecdotal - I've never seen a multi-hop traceroute show correctly, only single hop. This is in my 2 months of testing on iOS using various nodes in the field.

garthvh, 2024-03-01:
I tested this a few times and got maps for multiple hop trace routes.  I may need to make the timeframe longer than an hour.

closed 2024-03-01 by garthvh