meshtastic/Meshtastic-Apple #517: 🚀 [Feature Request]: Maintain/restore data between disconnects

tekstrand, 2024-02-23:
### OS

iOS, iPadOS, macOS

### Description

## Current behavior
When using a single client to connect to multiple devices, disconnecting from one device causes the data(messages, trace logs, etc) from that device to be lost.

## Expected behavior
The app should maintain the data from previously connected devices, and restore it on reconnect.

### Participation

- [ ] I am willing to submit a pull request for this issue.

thebentern, 2024-02-23:
The current behavior was actually deliberately implemented. It actually use to behave as proposed, but a lot of folks run multiple meshes (including myself) When you switch devices and it attempts to merge nodes, messages (tied to channels) from them, it's a disaster

tekstrand, 2024-02-23:
> attempts to merge nodes, messages (tied to channels) 

@thebentern was this a result of how data was stored on the client, or something else? I am not familiar with the data structures so just looking to understand where the conflict is.

thebentern, 2024-02-24:
> @thebentern was this a result of how data was stored on the client, or something else? I am not familiar with the data structures so just looking to understand where the conflict is.

The crux of the issue is that the nodes and messages coming in from one node can be from a completely different set of channels with different keys, lora channels, etc. When you pair to a node on a different mesh, those lists combine into a soup of nodes some of which are ones you can legitimately contact, some of which you cannot. It even lead to support issues where folks tried message stale node based on the entries from a previously paired device.

thebentern, 2024-02-24:
I don't hate the concept by the way. I think if we can ensure that the last nodes channel set is exactly the same as the one we're currently connecting to, it could work, but I don't know if the infrastructure in the datamodel is setup for that or not off-hand.

AssetBurned, 2024-02-24:
wouldn't it be an option to use the nodes ID as some form of container to store all relevant information in and just switch these containers when connecting to a different node?
that could also help with the situation that you want to connect a node to your iPhone when you are on the move and when you are at home to your Mac.
If those containers are stored on the iCloud accessible folder, then all the apps could access the dataset.

garthvh, 2024-02-24:
The messaging user is connected to the node, there is no reasonable way to blend together multiple different messaging users.

closed 2024-02-24 by garthvh

garthvh, 2024-02-24:
This is not a feature that is working anywhere, no feature requests have been created listings the huge number of things that would be required, no client has done any of the work required, and the firmware does not support a user switching between different nodes during the day.

AssetBurned, 2024-02-24:
technically the web client allows you to use different nodes and the associated information/settings.

garthvh, 2024-02-24:
> technically the web client allows you to use different nodes and the associated information/settings.

The web client has no persistent data store, that is the info pulled in off the node on connect.

tekstrand, 2024-02-24:
As a PWA though, it could have persistence. Could be a cool feature to add.

AssetBurned, 2024-02-25:
Which makes it even more interesting to store the config and dater in a separated silo for each node. Not only would that address the issue of having different histories per radio, but also would open the possibility for the app to handle multiple nodes at the same time (think of different frequencies for example).

garthvh, 2024-03-02:
The firmware would need a major re-write.  This is not a supported feature on any client currently, the user is connected to the node.

thebentern, 2024-03-02:
> @garthvh do you think it would make sense to add this to the 3.0 board for consideration?

We may already have an item about separating NodeInfo / User, which we've actually talked about as early as back in the 1.3.X era. If not, we need one out there.

AssetBurned, 2024-03-04:
I don’t know how much of that really depends on the firmware. I mean yes the known nodes are stored on the device itself, and so are the lates messages and the settings such as which channels and if MQTT is used… but ultimately all of that could be copied over to the iOS device and stored there. If the radio with the same identifier is connected , that set is loaded from the phones/ipads/macs storage. And then updated with the lates settings from the radio.
As far as I understood the messages are already stored on the Apple device.
Yes it would not address this ha as having one identity across multiple radios (like switching between a 433 and a 868 device) but it would address some of the issues already…. The work you do with having an “virtual” MQTT identity would go the next step.

AssetBurned, 2024-03-07:
So making the app capable of supporting multiple nodes/users/identities as an enhancementrequest.

garthvh, 2024-03-08:
The user is the node, why are you struggling with this concept so much?

AssetBurned, 2024-03-14:
Because a human being may want to use different actual radios to communicate with different groups of people, without loosing all the conversations?
Because the Apple Application is already interpreting content provided from the actual device that runs the Meshtastic firmware on it and therefore enhances (or artificially restricts) the abilities of those devices (e.g. longer message history, requirements for channel keys, changing channel names in case of camelcase, not allowing manual setting position data) and therefore not being just a keyboard and a larger screen.

Having a functionality within the app that detects that you are connecting to a different bluetooth device and then asking the user "hey if you connect to this new device, i have to reset the channel list and message history to match the new connection. Would you like me to save this so i can restore it when you connect to the device presently used?"
Is within the abilities of what this Application could do.