meshtastic/Meshtastic-Apple #522: 🚀 [Feature Request]: Highlight messages based on triggers

AssetBurned, 2024-02-24:
### OS

iOS, iPadOS, macOS

### Description

Currently the all messages are shown in the same style. And you can easily miss a chat where you get mentioned.

Therefore it would be good to have an option to highlight a message in which you got mentioned somehow e.g. in bold text.
To detect if you got mentioned „@<your shortname>“ and the User ID should be used.

also see https://github.com/meshtastic/Meshtastic-Apple/issues/514

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-02-24:
Yeah this is not twitter, there is no @ functionality built into the protocol.

closed 2024-02-24 by garthvh

AssetBurned, 2024-03-10:
Neither is selective deletion of messages or persistent storage of messages on the client device. Still this apps supports it ;-)