meshtastic/Meshtastic-Apple #524: 🚀 [Feature Request]: More Time Interval Selections for Position Broadcast Interval

jonathantdo, 2024-02-25:
### OS

iOS

### Description

When Smart Position broadcast is disabled, it defaults to a 15-minute interval for GPS connection updates. It would be beneficial to have options for shorter intervals like 30 seconds and 1 minute to provide more frequent updates, especially for users who turn off Smart Position.

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2024-02-25:
You should be using smart position if you want more frequent updates, broadcasting every 30 seconds is really bad for the mesh.

closed 2024-02-25 by jonathantdo

garthvh, 2024-02-25:
I did add 5 and ten minute options back

jonathantdo, 2024-02-25:
Thank you @garthvh!